Sinapse-Energia/AP-Sinapse #106: [Generic UART] 123; Configure UART not used by default

ralcaide, 2018-02-23:
[Config UART](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/MQTT-Sinapse-API#configure-uart)

sinapseenergia, 2018-02-26:
Note: 
Depending on HW configuration or model, some UARTs can be occuped by an already connected peripherial

In this case, after configuration request over already used UART, "ERROR" message should be send (Refered to Error Messages Table (This will be upload asap))

fpgamcu, 2018-03-09:
For now used 999; message if uart is busy or error appears.

ralcaide, 2018-03-22:
@fpgamcu , is this command developed outside of the engine? why?

fpgamcu, 2018-03-22:
While processing this command we need to set return message if fail or don't return anything if ok. For now we don't have possibility to manipulate return message from the engine in this way.
But now we can define in context text variable and use it for returning result. If it is ok to have message also when command succeed - I can change the code and move command to engine.

ralcaide, 2018-03-22:
Yes, please, insert it in the Engine returning always a result. Thank you