Sinapse-Energia/AP-Sinapse #107: [Generic UART] 323; - 723; Use UART in Generic way (On demmand)

ralcaide, 2018-02-23:
[Act over UART](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/MQTT-Sinapse-API#send-payload-through-uart-on-demmand)

[Response from UART](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/MQTT-Sinapse-API#generic-uart-information)

fpgamcu, 2018-03-09:
For now there is possible to use GPRS/WIFI UARTs with this command. Maybe it is good for debug. But should we have define to disable usage of in use UARTs by this command?

GonzSanch, 2018-04-11:
About UART's response. We connect an EN to a particle sensor with UART output. The output protocol of the sensor is the following: UART; baud rate: 9600, databits: 8, stopbits: 1, parity: no.

So we send this configuration: 130;1;9600;8;1;0;

After setting the UART, the next step it's send a 'Read particle measuring' action, that  code is: 0x68 0x01 0x04 0x93. Then I send: 323;1;68010493;HEX;0;

The expected response should be: 0x40 0x05 0x04 0xDF1 0xDF2 0xDF3 0xDF4 0x(Checksum)
example: 40 05 04 00 30 00 31 56

Instead of this the response I got its: 42 40 05 04

If I send an erroneous message the response should be: 0x96 0x96

But I get:  723;700035;1;42 96 96 42 4D;HEX;-1;-1;1354628917;
or
723;700035;1;42 96 96;HEX;-1;-1;1354628917;

In other test looked like it was printing the first byte from the last message. I will attach some screenshots here. I tried to send with an USB-TTL converter different messages in HEX and ASCII.

PD: Full Datasheet of the particle sensor: https://sensing.honeywell.com/honeywell-sensing-hpm-series-particle-sensors-datasheet-32322550-c-en.pdf

I set the EN with the sensor plugged in the External 3 for debug. 

[image]
[image]
[image]
[image]
[image]

ralcaide, 2018-04-17:
@fpgamcu , what is the status of this issue?. Could we have something soon?

fpgamcu, 2018-04-17:
In 2 hours I will finish with testing and we will have the solution.

ralcaide, 2018-04-18:
@fpgamcu , could you provide us feedback about the status, please? If you are not totally able to reproduce the problem, please tell us what do you need from our side. The fix of that is very necessary for us.

Problem: The sensor is sending several bytes but the AP is getting / returning only the first one

fpgamcu, 2018-04-18:
With second option MQTT commands will be the same, and it will be possible to receive variable length answers, but it needs about 3 hours to implement and check it.

ralcaide, 2018-04-18:
Ok, please go with the second option.

fpgamcu, 2018-04-18:
With last commit should work as expected.
By the way, if there will be no answer from device we would be able to indicate this only by timestamp. payload in 723; answer will be the same as in last successfully received package. is it ok?

ralcaide, 2018-04-18:
Yes, the payload is linked to the time-stamp, right?. It should be ok like that. We will test it asap.

GonzSanch, 2018-04-19:
Testing the new commit, in HEX now the mqtt response print only the firsts bytes of the response.

Seems like the response still saving the last message sended and print it the first time you ask for the response, so you have to ask 2 times to get a true response.

Attached screenshots and test review:

1.  Send an erroneous payload, should get **0x9696** response Neg ACK
Instead i get only 96  (first par)
[image]

2. Send Read measuring (68010493), should get (40 05 04 XX XX XX XX CHK), the answer is the last one received.
[image]

3. Send again Read measuring, then i got the first par of bytes from the true response (**40** 05 04 XX XX XX XX CHK):
[image]

To get extra info I will test with an USB-TTL and Docklight.

fpgamcu, 2018-04-19:
As I see - timestamp is 0 always - so you don't get response in all cases - device not answer or answer longer than GENERIC_UART_TIMEOUT time

GonzSanch, 2018-04-19:
The response message is the payload on black screen. The 0 parameter that you saw its _LOCATE_ on publish window, sorry if this has been a bit confused. 

As u can see after ;-1;-1; (LAT & LONG) parameters it's the timestamp, different in each screenshot.

I dont know which its the value to GENERIC_UART_TIMEOUT but the reponse time of the sensor its <6000ms ...

fpgamcu, 2018-04-19:
sorry, my fault :)
GENERIC_UART_TIMEOUT is 1000 msec, maybe better will be to increase for this kind of sensor. GENERIC_UART_TIMEOUT is defined in Definitions.h file.

ralcaide, 2018-04-19:
Ok, we will test with this value

GonzSanch, 2018-04-19:
Tested with an increment on GENERIC_UART_TIMEOUT (6000ms) and get the same result. 

[image]

fpgamcu, 2018-04-19:
Can you connect UART sniffer (RX and GND of USB-UART to TX and GND of sensor) to sensor in order to confirm that sensor sends more than one byte?

ralcaide, 2018-10-29:
Tested by @GonzSanch and passed.

closed 2018-10-29 by ralcaide