Sinapse-Energia/AP-Sinapse #108: [CMC - EN Integration] To integrate all the CMC features into Commboard-V2.0-F427VG

ralcaide, 2018-03-04:
To integrate all the CMC features into the branch of F427VG MCU. 
This MCU has more memory and normally should works also with the whole context.

fpgamcu, 2018-03-09:
Integration done, but there is possible issue. We have command 1; for RF/AP and command 1; for CMC. What will be context behavior it is have two same commands?

ralcaide, 2018-03-11:
This should not be a problem because the different 1; messages ares received through different topics.
If there is real problem we can change the id of the CMC command. @fpgamcu , what do you think?

fpgamcu, 2018-03-12:
Unfortunately current version of context engine don't have mechanism to check topic of command so for context there is no difference between this two 1; commands. Also you can notice that 313; command can be passed to INSTALLATION_NAME/AP_ID/ACT/EPD_ID instead of INSTALLATION_NAME/CMC/ACT/AP_ID and still will work, because of this context issue.

ralcaide, 2018-04-25:
If the only context to be used is the CMC I suppose will not have any problem. The problem will appears if we are using the whole context AP + CMC, right? 

I leave this issue in progress in order to find a solution

ralcaide, 2018-10-29:
Previous comment will be managed through other tasks

closed 2018-10-29 by ralcaide