Sinapse-Energia/AP-Sinapse #109: [Device ID - ALL] To use last 6 digits of the IMEI of transceiver as a Device_ID

ralcaide, 2018-03-06:
It is necessary to use the IMEI of the transceiver as a Device_ID (Last 6 digits). 
Now, Device_ID is hardcoded in the Context creation. This is not practical at all. Do the following:

- ETH connection: Use last 6 digits of the USR IMEI (Mac Address)
- GPRS connection: Use last 6 digits of the Quectel IMEI

fpgamcu, 2018-03-09:
As IMEI for ETH used its module ID, AT+MID. I don't fount in documentation AT command to get it's MAC. Maybe I missed something.

ralcaide, 2018-03-12:
Regarding the code, seems this  command is only available for ETH, isn,t? What is the status for GPRS?

fpgamcu, 2018-03-12:
For GPRS available too. Call to SetIMEI() was already in cmdflow after changes done in https://github.com/Sinapse-Energia/AP-Sinapse/issues/88

ralcaide, 2018-03-15:
Testing...

GPRS: OK
ETH : KO -> The ID displayed is the same set in the context. Tested with EEPROM

@fpgamcu, could you take to a look to this?

ralcaide, 2018-03-16:
@fpgamcu , are you able to reproduce the problem with the ETH ID in your device?

ralcaide, 2018-03-18:
@fpgamcu , could you please provide us a feedback about that?

fpgamcu, 2018-03-19:
The problem is AT+MID command: by datasheet it should return +ok=<module_id> but in real (at least in my board and I think in other also) return just "+ok=" and nothing more, so the ID is empty and FW keep the EPD ID same as in the context.
Please provide better replacement for AT+MID command that can be used as ETH ID.

ralcaide, 2018-03-19:
Ok thanks, we will ask to USR support.

ralcaide, 2018-04-27:
@fpgamcu , we should try to get an unique ID from USR gateway. Could you try a bit more with:

- AT+MID
or
- AT+REGID

If nothing works, please provide here the version of the FW.

ralcaide, 2018-04-27:
From USRIOT tell us to use: AT+WMAC. @fpgamcu , could you try, please?

ralcaide, 2018-04-29:
The previous command is not in the doc but should works

ralcaide, 2018-05-02:
Moved to current sprint. It should be finished asap

fpgamcu, 2018-05-02:
AT+WMAC works ok on my device

ralcaide, 2018-05-02:
Good news, I have question:

- How is the interaction with the task #120?

I suppose, at first, when the context is empty will be used the HW ID, then it is possible to change it using the command 209;. If the device is restarted, use the context ID (set by 209;). If the device is erased with 314; then will again the original data so the HW ID will be used. Right?

fpgamcu, 2018-05-02:
Now on any restart device will change it's ID to IMEI/MAC. I will fix it to keep ID if was set by 209;

ralcaide, 2018-05-14:
Tested OK with T230

closed 2018-05-14 by ralcaide