Sinapse-Energia/AP-Sinapse #110: [BUG][Actuation over Groups / Broadcast] The actuation over groups is not correct

ralcaide, 2018-03-14:
Acting over groups, the EN sends a lot of commands and the EPD ends always with status OFF. 

Acting directly over EPD: OK

RF TX 0D 2C 00 00 01 52 81 32 FF 01 A6 06 EB 
 RF TX 0D 2C 00 00 01 52 81 32 FF 01 A7 06 EC 
 RF TX 0D AC 00 00 01 52 81 32 FF 01 A4 06 69
 RF TX 0D AC 00 00 01 52 81 32 FF 01 A5 06 6A

Acting over Group FFFF07 : KO

RF TX 0C 22 00 00 01 FF FF 07 12 92 06 DE 
RF TX 0C 22 00 00 01 FF FF 07 12 93 06 DF 
**RF TX 0D 2C 00 00 01 FF FF 07 64 00 98 06 41**
RF TX 0D 22 00 00 01 FF FF 07 14 64 99 06 4C

**RF TX 0D 2C 00 00 01 FF FF 07 64 00 9A 06 43**
RF TX 0D 22 00 00 01 FF FF 07 14 64 9B 06 4E
RF TX 0D 2C 00 00 01 FF FF 07 FF 01 A0 06 E5
RF TX 0D 22 00 00 01 FF FF 07 14 FF A1 06 EF
RF TX 0D 2C 00 00 01 FF FF 07 FF 01 A2 06 E7 
RF TX 0D 22 00 00 01 FF FF 07 14 FF A3 06 F1 

RF TX 0C A2 00 00 01 FF FF 07 12 90 06 5C 
RF TX 0C A2 00 00 01 FF FF 07 12 91 06 5D

 **RF TX 0D AC 00 00 01 FF FF 07 64 00 94 06 BD** 
 RF TX 0D A2 00 00 01 FF FF 07 14 64 95 06 C8 
 **RF TX 0D AC 00 00 01 FF FF 07 64 00 96 06 BF** 
 RF TX 0D A2 00 00 01 FF FF 07 14 64 97 06 CA
 RF TX 0D AC 00 00 01 FF FF 07 FF 01 9C 06 61 
 RF TX 0D A2 00 00 01 FF FF 07 14 FF 9D 06 6B
 RF TX 0D AC 00 00 01 FF FF 07 FF 01 9E 06 63
 RF TX 0D A2 00 00 01 FF FF 07 14 FF 9F 06 6D 

Why are sended the commands in black?. These commands provokes turning OFF the lamp

fpgamcu, 2018-03-14:
Please check again with commit b04ce79

ralcaide, 2018-03-14:
Ok, could you integrate the change also into Commboard-V2.0 ? (This is the branch of F405VG)

ralcaide, 2018-03-15:
TESTED OK

closed 2018-03-15 by ralcaide