Sinapse-Energia/AP-Sinapse #113: [SD] The mounting of SD fails frequently

ralcaide, 2018-03-14:
We are testing several devices with SD card and they fails trying to mount the SD. The function f_mount returns always "FR_NOT_READY". It seems the problem is in the SDCardStart: 

`FRESULT res = f_mount(&SDFatFs, SDPath, 1);
	if(res != FR_OK)
		SDCardErrorHandler();`

What is the meaning of that? It could be a BUG? like:

https://community.st.com/thread/42224-sd-card-cannot-be-mounted-successfully-using-the-code-generated-by-cubemx

or 

https://stackoverflow.com/questions/40308519/sd-card-cant-read-write-stm32f4

ralcaide, 2018-03-14:
Please, use the external 1 in order to debug, if it is needed

fpgamcu, 2018-03-15:
It is definitely not a code bug because current code works ok with several SD card on my side, and as I remember works ok other board on your side.
I have debugged on external 1 and found next:
FatFS library stops at f_mount(), and checking more precisely at disk_initialize() inside f_mount() with next message:
/* Failed to initialize due to no medium or hard error */
So possible problems are next:
1) Board issue: bad soldered SD card socket or circuit problem, most possible variant
2) SD card problem, least possible variant
Point 2 we can easily check. Need to remove SD card from board and insert into external 1 PC, than check SD card with disk tool. If it is possible - please perform this actions and I will connect to external again to check SD card with disk tool.
To check point 2 you need to analyze SD card signals at socket side and at MCU side with scope.

ralcaide, 2018-03-15:
2. Tested, the SD card is OK

We are performing tests in new devices produced in a dedicated fabric, no prototypes any more. The HW is the last revision, reviewed by Eugene. Results:

**ETH (2 devices) - External 1**
- EEPROM : Works right
- SD : Does NOT work. FatFS library returns error FS_NOT_READY

**GPRS (1 device) - External 2**
- EEPROM : Is not mounted, not tested
- SD: Works OK (Except the 314; that is producing a failure)

HW: The design is the same for ETH / GPRS, everything looks fine.

FW: 
1. Could we tune the SD library in order to perform retries? 
2. Provide errors through MQTT? Now, if the SD fails, the system is restarted. Instead of that, only the SD Driver should be restarted but the MQTT connection should be performed and informs about SD errors
3. Modify some fatFS configuration options:

- SPI? 1 or 4 bits mode?
- Ignore card detect?
- Ignore write protect?
- Slow the speed?

For us is very strange that the problem is in the HW. This part of the board is very simple.

ralcaide, 2018-03-15:
[STM32 with SD Card and FatFs using CubeMX – Write error and write speed - Solution.pdf](https://github.com/Sinapse-Energia/AP-Sinapse/files/1816134/STM32.with.SD.Card.and.FatFs.using.CubeMX.Write.error.and.write.speed.-.Solution.pdf)

@fpgamcu , what do you thin about these problems / solutions? could be helpful for us?

ralcaide, 2018-03-15:
Could you give us a feedback about the configuration of the library? are we using DMA?

fpgamcu, 2018-03-15:
Yes, we using DMA.
First of all we need to be sure it is FW problem.
As I see we have 2 devices. On ETH device SD does not work, on GPRS device SD works. The FW is different because one compiled for ETH and another for GPRS. Lets compile ETH FW and flash it into GPRS device, it will not connect to GPRS neither to ETH but in debug we can validate SD operation. if ETH FW will work on GPRS device - than problem is in HW and we need to investigate more in that direction.

fpgamcu, 2018-03-15:
I just checked on GPRS (1 device) - External 2 FW with the same modifications as was on GPRS (1 device) - External 2.
So the FW is the same. Boards are different. On external 2 f_mount() return FS_NOT_READY, On external 1 it return FS_OK.
The problem is definitely HW. Best way to find HW problem will be to take both boards and check with oscilloscope signals on both of them, then find difference and investigate the reason.

fpgamcu, 2018-03-15:
About the link: we don't use cube generated code for fatfs. Instead we used code from STM32F4 reference example projects. But it should be almost the same.
About test with 15 boards, if boards will be the same with one on external 1 - they all can contain same problem. Without checking external 1 and external 2 boards with scope we will not get any progress. Even if you think that external 1 and external 2 boards are the same - they are not because same FW works different on them.
Did you performed any hardware verification on new boards, checked signals with the scope?

ralcaide, 2018-03-15:
Ok, thank you. We will try to study the signals.

Results of my tests:

Ext 1 (ETH) : Compiled for ETH = KO, Compiled for GPRS = KO
Ext 2 (GPRS) : Compiled for GPRS = OK, Compiled for ETH = OK

ralcaide, 2018-03-23:
After several tests, seems that was a HW problem. However there are still several SD problems to be managed through #114.

Closing this issue

closed 2018-03-23 by ralcaide