Sinapse-Energia/AP-Sinapse #114: [SD][Errors] Errors produced during SD use - Solutions

ralcaide, 2018-03-16:
During the testing of ENs GPRS and ETH it is appearing several errors. Try to understand them and fix them

*ETH*

Dev 1 (Ext 1): SD Not started (FS_NOT_READY)
Dev 2 (Ext 1): SD Not started (FS_NOT_READY)

*GPRS*

Dev 1 (Ext 2): OK
Dev 2 (Ext 6): SD Not Started (FS_NOT_READY)
287855 : OK
346248 : OK* , after several minutes, the error Node 346248 <16-3-2018 11:11:09> SDCardWrite: DEV.TXT create error 1 appers, then SD driver restarted and the device is not connected anymore.

ralcaide, 2018-03-16:
SD Errors with 287855 (is a .rar file) 

[SD_Error_287855.zip](https://github.com/Sinapse-Energia/AP-Sinapse/files/1819414/SD_Error_287855.zip)

ralcaide, 2018-03-21:
New error:

Node 700004 <21-3-2018 15:48:17> SDCardWrite: 57.SHD create error 1

@fpgamcu :
What is the meaning of 57? is the EPD number?
Any advancement about the SD stability?

ralcaide, 2018-03-22:
We are having big problems with the SD stability. It is necessary to find a solution providing stability. Questions / Possible solution:

- Are we using 4 wires or 1? : Please use 1
- Are we transmitting fast or slow: Please use the slowest possibility
- If there is a problem, the whole device is restarted or only the SD Driver? : Try to recuperate the system without restarting the whole device. Try to restart only the SD driver, then the SD card and of nothing works, then restart the system.

@fpgamcu , could you please give us a feedback about that?

fpgamcu, 2018-03-22:
Hi.
57 in 57.SHD is EPD number.
Now SD code works in next way:
For read - if read fails it just fails, SD driver will not be restarted.
For write - if write fails - restart SD driver. If done several (SD_RETRIES) times and no result - restart device.
We using 1 wire now and maximum possible speed (25 MHz as I remember).
I will modify the speed to lowest (400 kHz).

Unfortunately I was not able to reproduce the problem on my device, can I connect to some external in order to continue debugging?

ralcaide, 2018-03-22:
Thank you. Yes, you can use the external 2 that has a EN GPRS with SD. Current status:

- ID (IMEI) : 251646
- Topic base: TESTING_V1P2_SD
- 19 EPDs set up
- Periodic measurement each 10 minutes over the 19 EPDs

Please try there and give us feedback if you find something

ralcaide, 2018-03-22:
There are already errors:

Node 251646 <22-3-2018 11:40:34> SDCardWrite: 13.SHD create error 1

ralcaide, 2018-03-22:
Testing in external 2 with lower speed: uSdHandle.Init.ClockDiv            = SDIO_INIT_CLK_DIV;

A SD problem appears after 3 minutes:
Node 251646 <22-3-2018 14:02:54> SDCardWrite: 4.TXT create error 1

ralcaide, 2018-03-22:
More info about behaviour of external 2:

- If an error happens, then the system is always restarted but the SDCardStart fails always with a FS_NOT_READY
- After reprogram the device, the SDCardStart returns a FS_OK and continue working right until the next failure. The, it is necessary to reprogram the device in order to get it working again

ralcaide, 2018-03-22:
@fpgamcu, you have another external available with a EN V1P2 ETH with SD: External 6:

246912438 / napse1si

fpgamcu, 2018-03-23:
Test result after today morning external reboot:
Using external 2, topic TESTING_V1P2_SD, ID 251646
Run debug from TrueStudio
1wire SD 25 MHz
EPD list was clear
Set up 5 EPDs 300001..300005
Set up periodic actions for 300003 and 300005 to have Context switching and SD read/write actions
45 min test
No SD error
Set up more EPDs (with some real)
Set up one real EPD for periodic operation
20 min test
SD create error, restart, FS_NOT_READY
Log is in the attachments
[mqtt_sd.log](https://github.com/Sinapse-Energia/AP-Sinapse/files/1841065/mqtt_sd.log)
I think further debug should be made on the board with connected logical analyzer to see what is sent from and received by the MCU. Once I will receive board I will proceed with analyzer.

ralcaide, 2018-03-23:
I see here two different problems:

1. Why is the first error produced?. We know:

a. If there is not EPDs, the SD error does not appears
b. If we have few imaginary EPDs, the SD error seems not appearing
c. If we have several EPDs, the SD error appears always and is related to an EPD. 

Regarding this first problem: Is maybe related to the functions in charge of write / read the EPDs?

2. The SD is not restarted OK. But, if you reprogram the device, the SD works again. Is maybe related with the debugger mode?. Without debugger, the device is restarted and then works again.

fpgamcu, 2018-03-23:
I think point a and b should be confirmed in a long term test, at least 24 hours.
On point 2 I will investigate more.

ralcaide, 2018-03-23:
Regarding 1. Could you test also without RF in order to be sure that the problem is not an interaction with EPD writing when a RF command arrives?

Also, this post is very interesting:

https://community.st.com/thread/46310-solution-to-fix-stm32f4xxx-stmcubemx-firmware-v118-sdio-fatfs-dma-not-working-issue

Could you have a look?, maybe there is some valuable info.

Also, it seems he https://community.st.com/people/bardeenlai has a lot of knowledge. Maybe it is a good idea to ask him about our problems

fpgamcu, 2018-03-23:
sure, I will run the test, and check the comunity

ralcaide, 2018-03-27:
Questions about architecture:
1. Diagram Flow of the SD accesses. Is written when a RF message arrived? MQTT message creation? are we using some kind of buffer in order to save data before save it in the SD?
2. FreeRTOS tasks with priorities. Map of tasks / threads / priorities. Could you provide a list?
3.Configuration of the SD Stack?. Typical problem: "onfigMINIMAL_STACK_SIZE to uSDThread(). Disabling interrupts will avoid that the stack will be overwritten, and thus hide the problem."

ralcaide, 2018-03-27:
4. Shared use of the DMA. How is managed the use of the resources shared by different tasks?

fpgamcu, 2018-03-27:
1. SD access is almost always managed by the context engine (except saving non EPD device parameters once a minute in context daemon task). When RF message received there is no context operations, but when received RF message is processed - there is context operations. On mqtt message process is context operation. I can prepare context interaction diagram and analyzing it we can have SD access one.
2. I will provide map of tasks and OS objects (mutexes, etc).
3. Context (and SD) is used in several OS tasks: mqttReception (process MQTT message by engine), rfReception (process received RF message), rfSend (process TX/RX and signal counters), contextDaemon (save non EPD parameters once a minute), schedulerTriggers (scheduled operations).
4. DMA management is done by STM HAL layer.

fpgamcu, 2018-03-29:
Testing gives several important results:
1. On new hardware card detect pin is not working properly, checked in external2 and in new boards you send to our office - card detect works randomly. It works ok in previous comboard V1P1, at least in board that you send to our office before). I will check HW more today, but by the way I disabled card detect pin check.
2. FR_NOT_READY on startup and on SD restart was caused by this card detect issue.
3. Now when SD write or create error appears - SD card is restarted successfully. Please test more this last commit to be sure of device stability.

By the way, while testing I found also several reasons of SD error. Now with last commit this error are not fatal, but still I will investigate more and fix them.
1. Now FatFS is using RTOS heap and seems like lack of this heap is one of the reasons of SD errors. Possible solutions - increase RTOS heap or use generic heap. How was found: 
* disable context daemon task and using MQTT publish queue in scheduler task - no SD errors,
* enable context daemon task and leave RTOS heap for SD - SD errors appears,
* enable context daemon task and use generic heap for SD - no SD errors.
2. Using MQTT publish queue in scheduler task sometimes lead to SD errors. This need to be tested more. How was found:
* disable context daemon task and using MQTT publish queue in scheduler task - no SD errors,
* enable using MQTT publish queue in scheduler task - SD errors appears.

If you don't mind I will use external2 today to fix this 2 bugs.

ralcaide, 2018-03-29:
Thanks, in relation to the above explanations:

1. SD (card detect pin): I sent you the schematics, please let me know if you find something strange. In other way, how affect disabling this pin to the FW?. Will be not able to detect when the SD is not there, right?

2. FatFS and RTOS Heap: 
* I don't understand first test / result. Were you using the MQTT publish queue as memory for SD?
* RTOS Heap: Increasing RTOS Heap give better results? Cons of increasing this heap?
* Generic Heap: Cons of using this heap?

3. I don't understand the test context daemon - MQTT queue - scheduler task. I will call you in order to understand

ralcaide, 2018-03-29:
Yes, you can use external 2, of course

fpgamcu, 2018-03-29:
Sorry, maybe I explained ti wrong:
* disable context daemon task and disable disable using MQTT publish queue in scheduler task - no SD errors,

fpgamcu, 2018-03-29:
I am testing now heap options, once done I will describe results.

ralcaide, 2018-03-29:
Ok, but in your third test the context daemon and MQTT publish queue was enabled, right? And the SD error doesn't appear because of the use of the generic heap?

For me, it seems there are maybe several problems but I don't see them very clear.

Now, testing the last commit:

- 700004 / TESTING_V1P2 / 19 EPDS / Periodic measurement each 5 minutes

fpgamcu, 2018-03-29:
In all tests from point 1 using MQTT publish queue in scheduler task was disabled.
In tests from point 2 context daemon was disabled.

fpgamcu, 2018-03-29:
Maybe it is corrupting somehow the SD handlers in memory, or just taking all the heap (I don't think so, but still will check).

ralcaide, 2018-03-30:
Results of previous testing were more or less good. It is still appearing SD errors but the device is not restarted and the availability time is more acceptable. 

Now testing with last version 8d95fdc29e54e566d876944cfe75adb5680bf8e7

Same environment as previous tests:
- 700004 / TESTING_V1P2 / 19 EPDS / Periodic measurement each 3 minutes
- 700011 / TESTING_V1P2 / 19 EPDS / Periodic measurement each 5 minutes (previous version of the firmware: 5b5a7b3af5a9c4272975cf326050dfb48a7e26bf)

ralcaide, 2018-03-30:
Preliminary results after 5 hours testing:

**700004**

SD Errors : 0
Reconnections : 2

**700011**

SD Errors: 9
Reconnections: 6

It looks very promising!

ralcaide, 2018-03-31:
More results - 10:30 - after 17 Hours testing:

**700004**

SD Errors: 0
Reconnections : 11 , each reconnection takes around 1,5 - 2 minutes to be performed

**700011**

SD Errors : 23
Reconnections : 17

This device stop publishing after an SD Error:

`
2018-03-31 08:03:35 INFO Message: Node 700011 <31-3-2018 06:03:32> SDCardWrite: DEV.SHD create error 1 
2018-03-31 08:03:35 INFO Message: Node 700011 <31-3-2018 06:03:32> Restarting SD driver `

ralcaide, 2018-03-31:
Results at 20:30 - 27 Hours testing:

**700004**

SD Errors: 0
Reconnections : 12 , each reconnection takes around 1,5 - 2 minutes to be performed

Only one reconnection in last 10 hours

ralcaide, 2018-03-31:
Testing with a device with more EPDs (700006) are still appearing SD Errors.

ralcaide, 2018-03-31:
The device 700006 had the following characteristics:

- 120 EPDs
- Periodic Status each 15 minutes
- Several SD errors 
- After 2 hours the device is disconnected without sending any deafness message, neither error message.
- By the moment the device is not connected anymore

This fact is very problematic. Is the number of EPDs a problem?

ralcaide, 2018-04-01:
We think that some devices never connect again becuase a bootloader problem:

https://github.com/Sinapse-Energia/bootloader-stm32/issues/30

The devices that after some problem were restarted and never connected again have been recovered forcing a HW reset. The SD cards are OK

ralcaide, 2018-04-02:
The devices OFFLINE published as last message always: SD Driver Error. Then we think they stays in the bootloader (light OFF).
If the devices are restarted manually, connect again.

ralcaide, 2018-04-04:
Test results after last commit: 7d31b212a8fe8ed70d63d7c5769dfa8b7b82ad43

1. Device with 120 Virtual EPDs and 19 Real EPDs: SD error happens very frequent. The device sometimes stays in white led blinking and then is restarted. When is online, it seems that all the programmed actions are not executed anymore in a correct way (some EPDs are disappear). After several restarts the EN works very bad and not return the response to 313;. Solution: Format SD with the 314;

2. Tests performed with 10 EPDs : OK, 1 error SD
3. Tests performed with 30 EPDs : OK, 2 reconnections
4. Tests performed with 50 EPDs: TBD, 1 error SD but very strange because the error is in the EPD 85 but the EN have only 50 EPDs. @fpgamcu , any idea?

ralcaide, 2018-04-04:
With 40 EPDs seems that is working right. No SD errors after one hour.
With 50 EPDs, during one hour 3 errors were produced

ralcaide, 2018-04-05:
Results with 40 EPDs and periodic status publications:

- 13 Hours testing
- 7 SD Errors. Some related with a incoherent number of EPD, for example `SDCardWrite: 71.SHD create error`
- 3 reconnections
- Some published messages contains errors: `Incoming message #TESTING_V1P2/CMC/ACT/700011#:#316;100843; # `

ralcaide, 2018-04-05:
@fpgamcu, any idea about the previous comment? 
- SD error about an EPD not present?
- Message with an strange symbol. Memory error?

fpgamcu, 2018-04-13:
Testing results with commit 3e13fd
In general all looks good.
Got one reconnection because of ping at 01:35. No SD errors. 90 EPDs total. Ten EPDs is set for periodic 2;. Several EPDs is set to periodic 4; with period 3 and 4 minutes. No HF appear while testing.
Also during this test I noticed that signal from EPDs is much stronger (may be related with different way of receive processing).
[__TEST_external.log](https://github.com/Sinapse-Energia/AP-Sinapse/files/1906273/__TEST_external.log)

ralcaide, 2018-04-16:
Performing tests with the last bug fix we've found the following behavior:
`
13:02:47 16/4/2018:    714;TRUE; /TOPIC: TESTING_V1P2_INTEGRA/CMC/DEBUG
13:02:47 16/4/2018:    714;TRUE; /TOPIC: TESTING_V1P2_INTEGRA/CMC/DEBUG
13:03:04 16/4/2018:    313; /TOPIC: TESTING_V1P2_INTEGRA/CMC/ACT/700025
13:09:53 16/4/2018:    Node 700025 <16-4-2018 11:12:28> 700025 Connected(0) (CONTINGENCY) to soporte-tecnico.bitnamiapp.com, successfully subscripted to All topics /TOPIC: TESTING_V1P2_INTEGRA/CMC/DEBUG
13:10:24 16/4/2018:    Node 700025 <16-4-2018 11:12:58> SDCardWrite: DEV.SHD create error 1 /TOPIC: TESTING_V1P2_INTEGRA/CMC/DEBUG
13:10:24 16/4/2018:    Node 700025 <16-4-2018 11:12:58> Restarting SD driver /TOPIC: TESTING_V1P2_INTEGRA/CMC/DEBUG
13:10:29 16/4/2018:    313; /TOPIC: TESTING_V1P2_INTEGRA/CMC/ACT/700025
13:11:33 16/4/2018:    313; /TOPIC: TESTING_V1P2_INTEGRA/CMC/ACT/700025
13:19:32 16/4/2018:    Node 700025 <16-4-2018 11:22:07> 700025 Connected(0) to soporte-tecnico.bitnamiapp.com, successfully subscripted to All topics /TOPIC: TESTING_V1P2_INTEGRA/CMC/DEBUG
13:20:25 16/4/2018:    313; /TOPIC: TESTING_V1P2_INTEGRA/CMC/ACT/700025
13:20:31 16/4/2018:    313; /TOPIC: TESTING_V1P2_INTEGRA/CMC/ACT/700025
`

Then the LED was OFF, then green blinking, then OFF again, then green blinking, then OFF...
@fpgamcu , The 314;TRUE; was published two times as you can see. Any idea?

fpgamcu, 2018-04-16:
It is issue of last implementation. I have modified FW to send result twice. When sending result once sometimes it was not sent (function called but MQTT message was not received on PC side). In other case when sending result twice it is received always, but twice. I am still thinking what is the reason of this behavior, when last MQTT message before restart is not sent.

On other side, is SD erase works ok on this device? As I see device was restarted after erase, but then restarted again.

ralcaide, 2018-04-16:
After several reconnections seems that is working. The device was plugged ON always. Now is connected and working but seems that something strange happening.

It could be a problem to send 314; twice, without set up devices?

fpgamcu, 2018-04-16:
If you will send 314;0; twice it is not a problem, only first will be executed before restart.
For now when you send 314;0; as result you will receive 314;TRUE; twice. Unfortunately for now from FW I can either send 314;TRUE; once (but sometimes it is not send) or twice (guarantee sent, but always twice).