Sinapse-Energia/AP-Sinapse #115: [ETH][Deafness] The deafness is appearing in some circumstances

ralcaide, 2018-03-19:
It seems that the deafness is appearing again, in a real installation we have the following situation:

- EN V1p1 - 427 with bootloader
- FW : Comm1.1_427_V1.10
- EPDs : 80

After several days, the EN publish commands but does not response. The connection is not restarted.
TODO:

- Identify why is that happening
- Why the connection is not restarted? 
- In parallel, develop a security mechanism, restart the device daily, at a given hour provided in the Definitions.h.

ralcaide, 2018-03-19:
@fpgamcu , what do you think?

fpgamcu, 2018-03-19:
If situation is still same can you write ID and Topic here? I will check messages published.
Usually ping mechanism should restart connection, if not and ping debug messages displayed periodically - than receive task stuck somewhere, most likely at processing messages.
What is exact configuration of client FW: AP or CMC or Both? ETH or GPRS? Can you share Definitions.h file from client FW?

ralcaide, 2018-03-19:
Thanks for the feedback.
- Sadly, The device is restarted by hand and now is acting correctly. 
- AP ETH with IP Fix
- More info: Previous to deafness was sent several commands 3;; to several EPDs. Maybe is related. 

[Definitions_soleco.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/1824493/Definitions_soleco.txt)

fpgamcu, 2018-03-19:
Thanks, I will try to reproduce problem with this Definitions. Have you any communication log? If no - not a problem, I will try several commands including 3;

ralcaide, 2018-03-19:
Sorry but by the moment we don't have the communications log.
It seems that the deafness is appeared again. You can see it:

- soporte-tecnico.bitnamiapp.com:1883
- SOLECO/#

fpgamcu, 2018-03-19:
By now I don't receive anything on this topic

ralcaide, 2018-03-19:
Sorry, the client restarted the device. It will be not possible to use this topic in order to detect the problems. I will try to reproduce it in local

ralcaide, 2018-03-19:
Do not focus on this issue by the moment. I think the problem is related with the speed of the MQTT commands. The client sent several 3; without waiting time.

ralcaide, 2018-03-23:
The deafness appears when the AP receives several commands without space. By the moment we leave as it is. To be improved in next sprints