Sinapse-Energia/AP-Sinapse #118: [Bootloader configuration] Develop the procedure to manage the bootloader from client applicartion

ralcaide, 2018-03-22:
[Procedure ](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/Integration-with-Bootloader#when-should-the-bootloader-search-for-a-new-firmware)

Commands:

[207](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/MQTT-Sinapse-API#bootloader-configuration)
[208](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/MQTT-Sinapse-API#bootloader-activation)

fpgamcu, 2018-04-03:
Should we use for this settings same NVM type as client app?
Here can be problem with SD card libraries. Now we have 32K flash for bootloader and 28.4K is used by current bootloader code. I think we will not be able to include SD card libraries into bootloader because of size limitation.

fpgamcu, 2018-04-03:
According to STM32F427 flash map we can increase bootloader size to 64K, I think with this we will be able to use SD card NVM.
http://www.st.com/content/ccc/resource/technical/document/reference_manual/3d/6d/5a/66/b4/99/40/d4/DM00031020.pdf/files/DM00031020.pdf/jcr:content/translations/en.DM00031020.pdf

ralcaide, 2018-04-03:
No, it is not necessary to use the same NVM. But the EEPROM is not available in the GPRS model.
What about to use a little part of the Flash memory?. The variables to share are only very few and the mechanism to synchronize Bootloader and Client should work always without any kind of problem.
In other products we are using a part of the flash

fpgamcu, 2018-04-03:
Yes, using part of flash for this purpose is perfect.

fpgamcu, 2018-04-03:
I think 20 hours should be enough, including testing and debugging. I think I will have a version at April 5.

ralcaide, 2018-04-06:
Yes, 

URI: Generic HTTP URI
PATH: Network / HTTP path pattern. 

Anyway, these checks can be done through other issues in a next sprint. We leave this task without checks

ralcaide, 2018-04-25:
Tested in T203

closed 2018-04-25 by ralcaide