Sinapse-Energia/AP-Sinapse #122: [EEPROM][Group directions] EEPROM corrupted during actuation over groups

ralcaide, 2018-03-26:
In a real installation where we are using EEPROM instead of SD because SD is not stable we are finding that the EEPROM works right at first, sometimes during several days but once we act over group address (FFFFXX), the EEPROM problem appears. 

In this installation, the EN has between 100 and 150 devices around. 
How is managed the group address responses? Are processed? How can this be related with the EEPROM problem?

ralcaide, 2018-03-26:
@fpgamcu , I don't want a fix of that during this Sprint but please, give us feedback. Also maybe you have an idea about the possible problem

fpgamcu, 2018-03-27:
What kind of EEPROM problem appear? Do you have a communication log?
In current version of FW group addresses is located in context as EPD. Group requests are managed in special way, sending 2 request bursts, one for SE and one for EC. Group address responses now not managed in some special way. If we will receive response with group address in destination address field it will be processed according to general EPD logic, as EC response, if there is corresponding request in WR queue.