Sinapse-Energia/AP-Sinapse #123: [EXTRA] Improvement of Transparent Mode - No Context

ralcaide, 2018-03-27:
First part of transparent mode have been developed here: #87 

Due to the instability of the NVM it is necessary to improve the Transparent mode of the AP with the following characteristics and features:

1. If Transparent Mode is enabled. the EPDs subspace should be ignored. No operations related with EPDs
2. All the MQTT messages related with EPDs will have the same answer: `799;RF_FRAME;` in the topic `ID_INST/EPD/RF`. RF_FRAME is the the RF answer in HEX 
3. The RF responses should be saved in a queue and published each N seconds, configurable in the Definitions file.

The goal is to have a totally transparent RF gateway

ralcaide, 2018-03-27:
This task is on hold. The first option is to fix the SD Problems : #114 
If today we don't find a solution for SD, it should be implemented this new task. If we find a solution, then, the tasks stays in the backlog

@fpgamcu , what do you think about the complexity, feasibility and size (time) ?

fpgamcu, 2018-03-27:
I think size should be increased to 4, this task have middle complexity, but should be possible to implement is without the problems.