Sinapse-Energia/AP-Sinapse #124: [BUG][Party Mode] The device execute past commands in aleatory moments

ralcaide, 2018-03-28:
We observe that in some cases, the device execute commands sent in the past and normally already executed. Possible reasons:

- The messages are not deleted from the MQTT Received queue?
- The Broker is sending again the messages? MQTT connected with retain flag?

See the following log messages

Normal behaviour:

`15:14:02 27/3/2018: 3;100; /TOPIC: SOLECO/200011/ACT/FFFF02
15:14:02 27/3/2018: Node 200011 <27-3-2018 15:16:15> Incoming message #SOLECO/200011/ACT/FFFF02#:#3;100;# /TOPIC: SOLECO/CMC/DEBUG
15:14:29 27/3/2018: 3;40; /TOPIC: SOLECO/200011/ACT/FFFF02
15:14:29 27/3/2018: Node 200011 <27-3-2018 15:16:42> Incoming message #SOLECO/200011/ACT/FFFF02#:#3;40;# /TOPIC: SOLECO/CMC/DEBUG
15:15:35 27/3/2018: Node 200011 <27-3-2018 15:17:49> Sended Ping getting 1 /TOPIC: SOLECO/CMC/DEBUG
15:15:37 27/3/2018: 3;0; /TOPIC: SOLECO/200011/ACT/FFFF02
15:15:37 27/3/2018: Node 200011 <27-3-2018 15:17:51> Incoming message #SOLECO/200011/ACT/FFFF02#:#3;0;# /TOPIC: SOLECO/CMC/DEBUG
15:15:51 27/3/2018: 3;100; /TOPIC: SOLECO/200011/ACT/FFFF02
15:15:57 27/3/2018: Node 200011 <27-3-2018 15:18:11> Incoming message #SOLECO/200011/ACT/FFFF02#:#3;100;# /TOPIC: SOLECO/CMC/DEBUG
SOLECO/CMC/DEBUG`

MQTT reconnection (This part is not necessary for the Party Mode):

`15:30:15 27/3/2018: Node 200011 <27-3-2018 15:32:42> Sended Ping without getting PINGRESP (deafness?) /TOPIC: SOLECO/CMC/DEBUG
15:30:15 27/3/2018: Node 200011 <27-3-2018 15:32:42> Ready  to reconnexion or reboot!!!  /TOPIC: SOLECO/CMC/DEBUG
15:31:26 27/3/2018: 3;0; /TOPIC: SOLECO/200011/ACT/FFFF05
15:31:38 27/3/2018: Node 200011 <27-3-2018 15:33:18> 200011 Connected(4) to soporte-tecnico.bitnamiapp.com, successfully subscripted to All topics /TOPIC: SOLECO/CMC/DEBUG
15:31:38 27/3/2018: Node 200011 <27-3-2018 15:33:18> Hearing recovered after 1 tries!! /TOPIC: SOLECO/CMC/DEBUG`

Party Mode:

`15:31:43 27/3/2018: Node 200011 <27-3-2018 15:33:24> Incoming message #SOLECO/200011/ACT/FFFF02#:#3;100;# /TOPIC: SOLECO/CMC/DEBUG
15:31:57 27/3/2018: Node 200011 <27-3-2018 15:33:38> Incoming message #SOLECO/200011/ACT/FFFF02#:#3;40;# /TOPIC: SOLECO/CMC/DEBUG
15:32:17 27/3/2018: Node 200011 <27-3-2018 15:33:58> Incoming message #SOLECO/200011/ACT/FFFF02#:#3;0;# /TOPIC: SOLECO/CMC/DEBUG
15:32:31 27/3/2018: Node 200011 <27-3-2018 15:34:13> Incoming message #SOLECO/200011/ACT/FFFF02#:#3;100;# /TOPIC: SOLECO/CMC/DEBUG
15:32:45 27/3/2018: Node 200011 <27-3-2018 15:34:26> Incoming message #SOLECO/200011/ACT/FFFF02#:#3;30;# /TOPIC: SOLECO/CMC/DEBUG
15:33:05 27/3/2018: Node 200011 <27-3-2018 15:34:47> Incoming message #SOLECO/200011/ACT/FFFF02#:#3;0;# /TOPIC: SOLECO/CMC/DEBUG
15:33:19 27/3/2018: Node 200011 <27-3-2018 15:35:01> Incoming message #SOLECO/200011/ACT/FFFF01#:#3;100;# /TOPIC: SOLECO/CMC/DEBUG
15:33:33 27/3/2018: Node 200011 <27-3-2018 15:35:15> Incoming message #SOLECO/200011/ACT/FFFF01#:#3;30;# /TOPIC: SOLECO/CMC/DEBUG
15:33:46 27/3/2018: Node 200011 <27-3-2018 15:35:29> Incoming message #SOLECO/200011/ACT/FFFF01#:#3;30;# /TOPIC: SOLECO/CMC/DEBUG
15:34:07 27/3/2018: Node 200011 <27-3-2018 15:35:49> Incoming message #SOLECO/200011/ACT/FFFF07#:#3;100;# /TOPIC: SOLECO/CMC/DEBUG
15:34:21 27/3/2018: Node 200011 <27-3-2018 15:36:03> Incoming message #SOLECO/200011/ACT/FFFF07#:#3;40;# /TOPIC: SOLECO/CMC/DEBUG
15:34:23 27/3/2018: 3;0; /TOPIC: SOLECO/200011/ACT/FFFF05
15:34:35 27/3/2018: Node 200011 <27-3-2018 15:36:18> Incoming message #SOLECO/200011/ACT/FFFF07#:#3;0;# /TOPIC: SOLECO/CMC/DEBUG
15:34:48 27/3/2018: Node 200011 <27-3-2018 15:36:31> Incoming message #SOLECO/200011/ACT/FFFF03#:#3;0;# /TOPIC: SOLECO/CMC/DEBUG
15:35:02 27/3/2018: Node 200011 <27-3-2018 15:36:45> Incoming message #SOLECO/200011/ACT/FFFF03#:#3;30;# /TOPIC: SOLECO/CMC/DEBUG
15:35:22 27/3/2018: Node 200011 <27-3-2018 15:37:06> Incoming message #SOLECO/200011/ACT/FFFF07#:#3;0;# /TOPIC: SOLECO/CMC/DEBUG
15:35:36 27/3/2018: Node 200011 <27-3-2018 15:37:20> Incoming message #SOLECO/200011/ACT/FFFF04#:#3;0;# /TOPIC: SOLECO/CMC/DEBUG
15:35:50 27/3/2018: Node 200011 <27-3-2018 15:37:34> Incoming message #SOLECO/200011/ACT/FFFF04#:#3;100;# /TOPIC: SOLECO/CMC/DEBUG
15:36:10 27/3/2018: Node 200011 <27-3-2018 15:37:55> Incoming message #SOLECO/200011/ACT/FFFF04#:#3;0;# /TOPIC: SOLECO/CMC/DEBUG`

ralcaide, 2018-03-28:
@fpgamcu, any idea? 
We will try to reproduce in our installation

ralcaide, 2018-04-06:
This problem is still happening but we don't know from where is coming. It is very strange.

@fpgamcu , in order to try to understand, is it possible to publish the status of the MQTT queues periodically with the ping message? It should be great to have the number of messages in each queue

I think this is very fast to implement and helpful

ralcaide, 2018-04-06:
@fpgamcu , by the moment this is not necessary. Don't work on it, please.

fpgamcu, 2018-04-06:
Yes, it is possible to publish status of MQTT send-from-client queue.
For MQTT receive-to-client queue it is possible too, but will be a little complicated.
Because in send-from-client queue we have separated strings objects and in receive-to-client queue we have byte stream of not parsed MQTT messages with all its headers.

ralcaide, 2018-04-06:
Thanks for the feedback. In fact, we are very lost about the problem. The EN execute messages that were sent in the past, in some moment. We were thinking about the retained flag of the MQTT, if a client send a message with the retained flag enabled to a topic, any client subscribing to this topic will receive these messages (only the last one) automatically from the broker when is connected. After several tests seems this is not the problem. In fact, the are two possibilities:

- The broker send messages saved somehow because a MQTT configuration
- The client (our application) execute some past messages saved in a queue

It is necessary to say that the client is publishing the Debug message telling that a MQTT message was received. 

In order to understand the problem it could very helpful to add some extra information to the debug message that is displaying the reception of a message. This info can be the level of the reception queue but the most valuable info could be the MQTT Header of the message that contains some valuable data like the retain flag, QoS of the message, etc. 
@fpgamcu , what do you think about that? How can be the complexity of add these values to the debug message.

fpgamcu, 2018-04-09:
Existing Debug messages:

Message|Description
----------|------------
Incoming message #\<topic\>#:#\<message\>#|Published when MQTT reception task receive MQTT message
RECONNECTED because of communication breakdown after \<n\> tries!!!!|Published when MQTT send task reconnect to broker because can not send message
Reboot!|Published from scheduler thread when reboot was triggered by MQTT command (in real not published, seems like USR or M95 need some time to publish, but we reboot right after trying to send message)
Sended Ping without getting PINGRESP (deafness?)|Published when scheduler thread send ping but don't get response
Ready  to reconnexion or reboot!!!|Published before reconnection after previous message
Hearing recovered after \<n\> tries!!|Published after reconnection after previous message
Sended Ping getting \<n\>|Published after successful ping
Go to DISCONNECT from this server and connect to \<host\>:\<port\>. Bye|Published from scheduler thread when communication parameters is changed by MQTT command
This has been a RECONNECTION!!!|Published after previous message after reconnection was done

Added debug messages:
(MQTT Queue messages added not into Ping message, because Ping is executing only if there is no messages in queue, so putting this debug messages into ping is quite useless)

Message|Description
----------|------------
MQTT send queue size \<size\> strings, \<msgs\> messages|Published from scheduler task each 15 sec, MQTT send queue size in strings and in messages (each message is 2 strings, one for topic and one for message)
MQTT recv queue size \<size\> bytes|Published from scheduler task each 15 sec, MQTT receive queue size in bytes
MQTT ctrl queue size \<size\> bytes|Published from scheduler task each 15 sec, MQTT control receive (ping, connect, etc.) queue size in bytes
MQTT RX \<byte1\> \<byte2\>|Published when MQTT reception task receive MQTT message, contain message RAW bytes

ralcaide, 2018-04-10:
Thank you for the verbose debug. In fact, now it is very difficult to understand the debug information because there are a lot of info. 
Is it possible this special MQTT debug info a particular topic like:

ID_INST/ID_EN/MQTT_DEBUG
?

We are testing with several ENs and will be very useful to have a topic per EN

fpgamcu, 2018-04-10:
yes, sure, it is possible. I will implement