Sinapse-Energia/AP-Sinapse #125: [EN - 4;] To improve the period management

ralcaide, 2018-04-03:
This task is related with #67.

To modify the management of the publication period in order to have a any kind of 24 hours period:

T1 and T2 where both are between 00:00 and 23:59
Publications will be done between T1 and T2, even if are from different days
T1 = T2 and NTIME = -1, publications are done always

The message 4;5;22:00;06:00;-1; provokes publication between 22:00 of day D and 06:00 of day D+1
The message 4;5;04:00;16:00;-1; provokes publication between 04:00 of day D and 16:00 of day D
The message 4;5;04:00;04:00;-1; provokes a constant publication of the status each 5 minutes

fpgamcu, 2018-04-16:
Estimation is 4 hours, I will do it

ralcaide, 2018-04-25:
Tested in T224

soporteHW, 2018-10-22:
It's necessary to retest this feature using the improve for L3_READ_METERING_P;
[API definition](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/MQTT-Sinapse-API#ask-about-status-periodically)

soporteHW, 2018-10-23:
After sending the command "L3_READ_METERING_P;5;16:15;08:00;-1;" , device **doesn't publish** PERIODIC messages.

[image]

If the message sended is "L3_READ_METERING_P;1;16:15;20:00;-1;", device publishes PERIODIC messages every 1 minute.

[image]