Sinapse-Energia/AP-Sinapse #126: [BUG][Scheduled Actuation - 2;] The scheduled actuations provokes reconnections - restart if act over more than 2 EPDs

ralcaide, 2018-04-05:
Testing with  `2;100;12:10;80;12:20;60;12:30;40;12:40;20;12:50;0;13:00;50;13:10;100;13:20;20;13:40;` in a EN with SD over:

1 EPD : OK
2 EPDs: Sometimes the device is reconnected at the first step time (12:10) and the scheduler is not executed
3 EPDs: The device is **always** reconnected at the first step time (12:10) and the scheduler is not executed

Seems there is a problem with RTOS - Context - Memory

ralcaide, 2018-04-05:
Maybe related with #114

ralcaide, 2018-04-10:
Normally, the engine allows 10 steps maximum but if there are less steps it execute only the provided ones. 
This is working like that in the original engine. I think in the AP also.

Anyway, with 1 EPD is working good, with 3 don't

fpgamcu, 2018-04-10:
I have tested with 1, 2 and 3 EPDs with next conditions:
6 EPDs total, EPDs for test 300001, 300002, 300003 (not present / no answer)
problem not found.

How many total EPDs you have when testing? What EPD IDs was used for testing (1, 2 and 3)? Was it EC or SE?

ralcaide, 2018-04-10:
514853 (60 W)
514856 (60 W)
514583 (60 W)
100181 (35 W)
529333 (150 W)
140279 (40 W)
140285 (40 W)
140439 (40 W)
140267 (40 W)
140371 (40 W)
140435 (40 W)

ralcaide, 2018-04-17:
This issue will be used to study the MANY EPDs problems.
After testing we've found the following problem:

- 40 EPDs, scheduled status each 5 minutes: The status is published correctly **but** each 10 minutes
- 50 EPDs, scheduled status each 5 minutes: The status is never published. 

The execution of RF questions is taking more than one minute and the publications of MQTT messages is skipped. 

Solutions proposed by @fpgamcu : 

> 1) Maybe better will be not to split all into 2 iterations, but perform all at once. Ask RF, wait 2-3 seconds after last RF ask in current iteration, then send MQTT.
For example for 100 EPDs scheduled for each 10 minutes it will be something like this: Ask 100 EPDs RF (and while asking we receiving answers so RF answers will not be lost), wait 2-3 sec after last ask to receive last answers, send MQTT for 100 EPDs. Like this we would have guaranteed RF actions for 100 EPDs.
One issue I see here is when we will have something like scheduled actions for 50 devices each 10 minutes and for 50 devices each 11 minutes last 50 actions will not be performed. One solution is to have granularity 10 minutes, so period can be send only to 10, 20, 30 minutes.

> 2) Other, more complicated solution is flagging system. For example one simple RTOS task only check timings and set "action" flag when time is come for periodic action. Other task is checking flags and perform action when "action" flag is set. Like this actions will be performed any way, because time check is performed in different task not held by action itself.

> 2.1) Most right way is of course having parallel access to EPDs in context engine, because now when one task operating with EPD other tasks should wait till it end operation, it is because we can not switch to other EPD till finish operation with previous one. So I should remember to not to have in code operation with one EPD longer that one minute because it can lead to scheduler skips.

> I think first solution is fastest, but limits us to 10 minutes scheduler granularity in many EPDs system (I will implement and test it tonight because it is very fast to implement). Second solution is more closer to "right" implementation but need more time to implement and test it.

I think the correct roadmap at this moment is:

1. Implement solution 1 (Limitations)
2. Allow access of N EPDs in paralel? 
3. Implement proposed solution 2

fpgamcu, 2018-04-17:
Implemented solution 1. Current limitations:
Periods and time in periodic actions should be multiply of 10.
Example periods: 10, 20, 30 minutes.
Example time: 11:10, 11:20, 11:30.
Example 2; command: 2;10;06:20;11:06:30;20;06:40;0;21:10;......
Example 4; command: 4;10;00:00;23:50;-1;

fpgamcu, 2018-04-17:
Another limitation: if action is scheduled for 11:10 it could be executed at 11:10-11:19.

fpgamcu, 2018-04-17:
After last commit this limitation:
> if action is scheduled for 11:10 it could be executed at 11:10-11:19  

would almost not appear.
Most of the time actions scheduled for example to 11:10 would be executed as closer to 11:10 as possible.

ralcaide, 2018-04-18:
Thanks for these improvements. 
We saw an effect that I don't like during the testing of the previous commits.

Set up 90 Virtual EPDs and two of them have "real" values:

300133:  100;700025;300133;28;1;97;166.5;234.0;39.0;0.0;0;0.0;0;0.0;0;1355252616;
300136:  100;700025;300136;28;1;100;183.3;236.0;43.2;0.0;0;0.0;0;0.0;0;1355252622;

Then after set an scheduled action (2;) we receive the normal basic values. Everything zero and V = 220.

How is that possible?

fpgamcu, 2018-04-18:
is timestamp also changed during receive of 2; results?

ralcaide, 2018-04-18:
No, after that the timestamp was 0. Explanation of the tester:

> To make it clear: i used script to add 50 epds to the memory and to retrieve statuses from them (msg 4;10), I observed for 20 minutes - i got all the statuses twice, at 12:00 and 12:10, and in those cases this two virtual epds had that different statuses. Then I used script to set a scheduler for all of them (2;) and when i received new statues at 12:20 all of the epds had the same statuses, all 0 and one 220
and in those 2 cases 12:00 and 12:10 they looked the same, the same timestamps too

Maybe is not a big deal but maybe you have some clue

ralcaide, 2018-04-18:
New feedback from tester:

> 100;700025;300180;36;1;99;166.5;238.0;39.6;0.0;0;0.0;0;0.0;0;1355256264;`
100;700025;300181;34;1;100;183.3;236.0;43.2;0.0;0;0.0;0;0.0;0;1355256270
i got two weird statuses again
scheduler is set up, but I didnt set the time now. in the device is 15:00 and dimming should start at 12;10

It seems, some EPDs are updated even the wasn't any RF response to it

ralcaide, 2018-04-23:
No, the timestamp is always the same but is still strange. When we setup for example 300 EPDs, almost all of them have "real" values.

I think this is an bug, isn't?

fpgamcu, 2018-04-23:
When EPD is created in context there is only several values initialized (schedulers, PNOM, etc.). I think for other values context engine pick random something, maybe from other EPDs. Maybe would be better to initialize all the variables of EPD when one created? If yes, I will implement that, it will not take a lot of time :)
Should I create table in Wiki with default values for context variables?

ralcaide, 2018-05-03:
Move it to backlog. Will be developed in next sprints

ralcaide, 2018-05-07:
The implemented solution 1, provides more stability but the devices still reconnects a lot. I think is necessary to develop the solution 2.
What is the estimation @fpgamcu ? Could you develop that in the current sprint (following next two weeks)?

fpgamcu, 2018-05-07:
Estimation is about 8 hours for flagging system (development and test).

ralcaide, 2018-05-07:
Ok, please perform it after the other tasks of this sprint because are smaller tasks

fpgamcu, 2018-05-15:
Almost done, I will push the code tomorrow morning.

fpgamcu, 2018-05-21:
Flagging system implemented, but there is small issue - it will not give any advantage comparing to previous scheduled until parallel context operations will be allowed.

ralcaide, 2018-05-22:
This is a bad new. What exactly means "parallel context operations"?

fpgamcu, 2018-05-22:
Parallel context operation is write parameter or one EPD while working with other.
New our operation is next:
1) lock context
2) set EPD ID
3) write/read parameters, work with RF, etc.
4) unlock EPD ID

The problem is next - while executing scheduler action for one EPD we can not set flag for another EPD. So flagging system is useless because it waits for other operations with context and actually is blocked during those operations.

ralcaide, 2018-06-25:
After speak with the Engine Developers seems that there is not a viable solution at this moment.

Proposal:

To create an "event queue" and after finish the operations of the current EPDs set flags in the required EPDs.

@fpgamcu , what do you think?

ralcaide, 2018-07-10:
@fpgamcu , any feedback?

ralcaide, 2018-10-29:
@fpgamcu , status of this task?

ralcaide, 2018-11-30:
Let me think about that and I will provide you feedback. Now this task is paused