Sinapse-Energia/AP-Sinapse #131: [RF RX] To improve the management of RF responses

ralcaide, 2018-04-11:
The process of RF responses should be improved in order to increase the performance of our device

**Now**

All the RF responses raise the RX flag and the FW save the message in the RF_RX queue. Then, if the message is not valid or not related with an EPD of the EN is ignored.

This mechanism consumes a lot of resources for RF responses that will be ignored

**Proposed improvement**
- Try to raise the RX flag only if the message will be processed. Maybe it is not possible, then:
- Save the RF message in the queue only if will be processed.

A message will be processed by the EN if:
- The destination address is an EPD available in the NVM
- The message contains an id that the EN is able to process. For example, for ECs we process by the moment only the 0x01 (Coverage) and the 0x09 (Load)

fpgamcu, 2018-04-12:
1. yes
2. no, it was not processed. When processed is done you will have next messages:
Process_RF_RECV_queue: 1 messages
Process_RF_RECV_queue 0D 29 50 25 72 00 00 01 1F 01 0B 06 4F 
Process_RF_RECV_queue origin valid
Process_RF_RECV_queue destination valid
Process_RF_RECV_queue format valid
Process_RF_RECV_queue checking WR, cnt=2
Process_RF_RECV_queue WR valid

ralcaide, 2018-04-13:
2. Ok, but the frame is saved in the queue or not?. All the frames with origin 000001 (Any EN) should be ignored as soon as possible

ralcaide, 2018-04-25:
Tested in T222

closed 2018-04-25 by ralcaide