Sinapse-Energia/AP-Sinapse #133: [Digital Sensors L1] Integrate Digital Sensors SB functions and Develop the L1 MQTT API messages

ralcaide, 2018-04-24:
Integrate the SB functions developed here:

https://github.com/Sinapse-Energia/M2M-DigitalSensors-Development

And develop the L1 MQTT functions:

To be explained by RAE

ralcaide, 2018-05-02:
The main.c file of the attached github project contains an example of use of the Sensor SHT1 in order to get the Temperature and Humidity

soporteHW, 2018-05-02:
The SHT10 sensor is connected on EasyNet as follow (only in this case for test and integration sensor, in other case the pinout could be different)

- SCK_signal: pin PB6. This pin is shared with I2C1_SCL, so I2C1 interface must not be in use.
- SDO_signal: pin PB7. This pin is shared with I2C1_DA, so I2C1 interface must not be in use.
- CS_signal: PE6. This pin is unused but it's necessary to define it.
- SDI_signal: PE7. This pin is unused but it's necessary to define it.

Ask me if you have doubts.