Sinapse-Energia/AP-Sinapse #135: [L3 - Temperature and Humidity - On demmand] Develop MQTT L3 commands in order to get temperature and humidity from SHT1 sensor

ralcaide, 2018-05-02:
Develop the MQTT L3 commands in order to get the temperature and humidity from a sensor. By the moment the sensor to be used is the SHT1 but the sensor to use should be easy to modify in compilation time (Through a DEFINE)

**MQTT read Temperature and Humidity on demmand**

Question

Topic: `ID_INST/CMC/ACT/ID_AP` 
MSG: `L3_OD_TH;`

Response 

Topic: `ID_INST/CMC/MEASUREMENTS`
MSG: `L3_OD_TH_R;TEMPERATURE;HUMIDITY;TIMESTAMP;`

Task related with #133

soporteHW, 2018-05-14:
Test done. Sensor SHT1 integration works fine.

Only a little BUG was found. TIMESTAMP param in MQTT msg response is wrong (sww attached image)
[image]

1357567428 equals 01/07/2013 @ 2:03pm (UTC)!!

fpgamcu, 2018-05-15:
Thanks for the feedback, I will check the timestamp.

soporteHW, 2018-05-17:
Another BUG found. The MQTT message reponse doesn't include the AP_ID to identify the device.

ralcaide, 2018-05-17:
This is not a bug, is lack in the definition. 
I,ve modified the description of the issue in order to set add the AP_ID in the response message.

@fpgamcu , could you perform this little modification as soon as possible, please?. I think is very easy and fast, isn't?

soporteHW, 2018-05-23:
Once restested this issue, the AP_ID now is included in the response message.
However, the TIMESTAMP is still bad.

Attached an image:

[image]

soporteHW, 2018-10-23:
TIMESTAMP value fixed in BL V1.30.0 and CL V1.34

closed 2018-10-23 by soporteHW