Sinapse-Energia/AP-Sinapse #137: [Context Variables] To init the EPD variables to a default values

ralcaide, 2018-05-07:
Now, some EPDs variables (when there are a lot of EPDs) are taking some random values, see #126.

To init the values of an EPD to default values. All zero except the ID. V should be also zero, this will show that the EPD never received a valid RF

ralcaide, 2018-05-22:
To be tested through the Virtual EPD steps of C67

soporteHW, 2018-05-25:
There is still an EPD with random values. The rest send all zero except the ID.

[image]

On the other hand when you send a command 313; it just says there are 91 EPDs instead of 100.

[image]

fpgamcu, 2018-05-25:
On last screenshot there is 101 EPDs count, not 91 (the first number after ID), but indeed displayed only 91 (and the last is not full). I will check this in #43 issue
Need to the code that set variables to 0, seems like context engine bug.

soporteHW, 2018-05-25:
Theoretically there should only be 100 EPDs, the SD was cleaned and a script was sent to set up 100 EPDs, from 100101 to 100200.  I suppose the data after the ID should be 100 too, no?

fpgamcu, 2018-05-25:
yes, should be 100.

ralcaide, 2018-06-23:
After test again, the number of EPDs is correct. 

The values of some EPDs are still different to zero even if they are virtual. This fact is causing very strange problems in the integration with other systems because seems that the EPD has correct data received at least once. 
@fpgamcu , please could you take a look again? If it is an Engine problem let me know in order to speak with the Engine developers.

fpgamcu, 2018-06-25:
ok, I will check it

ralcaide, 2018-06-25:
Please, this behaviour should be fixed asap. If it is an Engine problem, let me know

fpgamcu, 2018-07-24:
Tested with several virtual EPDs and got all values 0 in answer to 1; and 4;