Sinapse-Energia/AP-Sinapse #138: [Retrieve FW Version - 209;]  Ease the retrieving of the FW version

ralcaide, 2018-05-07:
Now, it is possible to retrieve the FW version using the commands 207; or 208; but these commands have other implications. Develop the command 209; in order to retrieve the FW version in an easy way:

[Retrieve FW version](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/MQTT-Sinapse-API#retrieve-fw-version)

ralcaide, 2018-05-07:
Because of this command is necessary to rename the Customization of the API: #120

ralcaide, 2018-05-22:
To be tested through C133

soporteHW, 2018-05-24:
The FW version wasn't sended
[image]

fpgamcu, 2018-05-24:
It was sent, but it is empty, since we do not have code anywhere to set it to some value.

ralcaide, 2018-05-24:
This version should be set in the context, is not there?

fpgamcu, 2018-05-24:
It is set to the context from shared with bootloader data, but in those data it is empty (bootloader is initializing those data on first time)

ralcaide, 2018-06-12:
We are not able to update already installed bootloaders. It will be great to set the FW version from the client if it is empty. Also, the FW version in linked to the Client, not the bootloader. 
Is it possible to modify that, please?

fpgamcu, 2018-06-13:
Yes, sure. I can add a code that will check the FW version in shred memory and set it to current FW version, if different. What format should be used for FW version, the same as the tag in the github?

ralcaide, 2018-10-29:
@fpgamcu , status of this task?

soporteHW, 2018-11-05:
Tested passed

**EN device**
[image]

**AP device**
[image]

closed 2018-11-05 by soporteHW