Sinapse-Energia/AP-Sinapse #140: [RF Performance] The RF performance decreases after several days working

ralcaide, 2018-05-20:
The RF performance decreases after several days working, in some cases the AP stop receiving RF.
After restart with the 314;0; the performance is "good" again and then, day after day is getting worse. 

It is necessary to have a more stable RF Behaviour

ralcaide, 2018-05-20:
@fpgamcu , any idea?

ralcaide, 2018-05-28:
@fpgamcu , have you tested? is reproducible in your place? 
This is a huge problem for us at this moment. Seems that after restart (398; or 314;) the device improves the RF result but not enough.

could you please treat this task with high priority and provide us feedback?

fpgamcu, 2018-05-28:
I tested some time, but was not able to reproduce. Ok, I will do this in high priority. Now I placed device on another test, increased RF activity. Hope I will be able to reproduce problem.
btw, 
>Seems that after restart (398; or 314;) the device improves the RF result but not enough.

So device is not totally recovered from this bug after restart?

ralcaide, 2018-05-28:
We got RF responses but seems the performance is lower that at the beginning

ralcaide, 2018-06-11:
@fpgamcu , any idea about that? have you performed the RF spectrum research?