Sinapse-Energia/AP-Sinapse #141: [CMC Board compatibility] To check the CMC code is compatible to current code

ralcaide, 2018-05-21:
Is necessary to check the current code is compatible with the new CMC Board

ralcaide, 2018-05-21:
@francisjjp , upload here the HW-FW document

francisjjp, 2018-05-21:
[[HW-FW]CMC_BB_[V4p0]_Rev1.1.pdf](https://github.com/Sinapse-Energia/AP-Sinapse/files/2022238/HW-FW.CMC_BB_.V4p0._Rev1.1.pdf)

Here, is attached the HW-FW document for driving one CMC_BB board using the CMC_CB board.

francisjjp, 2018-05-21:
@fpgamcu, @ralcaide .

 As you can see in HW-FW document. There are following changes:

**RELAYS:**
 The seven relays are tied to other GPIO PIN. You can see in final table the mapping of GPIO.

 **EXPANSOR GPIO:**

New feature to implement. This is a expansor GPIO pin. We want to set three bits (FREE OUTPUT GPIOx) and getting information from other three bits (FREE INPUT GPIOx). The expansor IC is connected to microcontroller 427VG throughput I2C1 interface (PB6, PB7)

The getting of bit information and setting of bit information is related directly with one specific MQTT command.

ralcaide, 2018-05-21:
The Expansor GPIO feature will be not developed by the moment.
It is necessary only to check that the current FW is compatible with the new HW. If it is not compatible, new tasks will be created in order to adapt it in a correct way

fpgamcu, 2018-05-22:
It is an easy task - maximum an hour.

ralcaide, 2018-05-22:
Please do that with high priority, like that we are able to test the new hardware

fpgamcu, 2018-05-23:
Sorry, took more time because PWM and ADC pins also was changed

fpgamcu, 2018-05-23:
Here is binary generated for current code version.
[AP-Sinapse.binary.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/2029949/AP-Sinapse.binary.txt)

fpgamcu, 2018-05-25:
There was a bug in previous code in CMC variables definition.
Here is the fixed binary (CRC already added at the end).
[AP-Sinapse.binary.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/2038980/AP-Sinapse.binary.txt)

francisjjp, 2018-05-25:
@fpgamcu . It has been tested relays, but there are some errors:

**Bug1)** 

GPIO_RELAY1_Pin must connected to PC3, and not to PC4

**Bug2)**

Because of the relays start from number 0 and the  number is 6, into wrsouth.cpp file, the functions that
enabling/disabling relays goes from write_relay1 to write_relay7. I mean...

for first relay that is mapped to GPIO_RELAY0_Pin, the function that is used is write_relay1, then the final switching is over second relay and not over the first one. There are a shifting of one relay to right.

fpgamcu, 2018-05-25:
Thanks for the feedback. I will fix it. For Bug 2 If I understand right RELAY 0 in doc should become RELAY 1 in wrsouth, RELAY 1 in doc - RELAY 2 in wrsouth, etc ?

francisjjp, 2018-05-25:
Regarding bug2). I watch that if I send one MQTT command for activating one relay, next relay is activated. I suppose is because this the matching of Write_RelayX() with the definition of GPIO_RELAY1_Pin.

Now we have following functions. 

char *Write_Relay**X**(const char *value)
{
#ifdef GPIO_RELAY**X**_Pin
	GPIO_PinState state = GPIO_PIN_RESET;
	// If 1 or 0 - change pin state
	if (ParseRelayCmd(value, &state))
		HAL_GPIO_WritePin(GPIO_RELAY**X**_GPIO_Port, GPIO_RELAY**X**_Pin, state);
#endif
	return (char*) value;
}

We must set this:

char *Write_Relay**X**(const char *value)
{
#ifdef GPIO_RELAY**X-1**_Pin
	GPIO_PinState state = GPIO_PIN_RESET;
	// If 1 or 0 - change pin state
	if (ParseRelayCmd(value, &state))
		HAL_GPIO_WritePin(GPIO_RELAY**X-1**_GPIO_Port, GPIO_RELAY**X-1**_Pin, state);
#endif
	return (char*) value;
}

francisjjp, 2018-05-25:
Or maybe, you can fix in another way. Changing the mapping in main.h.

GonzSanch, 2018-05-25:
about: "Thanks for the feedback. I will fix it. For Bug 2 If I understand right RELAY 0 in doc should become RELAY 1 in wrsouth, RELAY 1 in doc - RELAY 2 in wrsouth, etc ?"

The problem comes when in main.h, You define {GPIO_RELAY0_Pin, GPIO_RELAY6_Pin} meanwhile in wrsouth Write_Relay its calling for {GPIO_RELAY1_pin, GPIO_RELAY7_Pin}, and #def GPIO_RELAY0_Pin7 its not define in main.

Hope it helps you!

fpgamcu, 2018-05-25:
Thanks, yes I noticed when implementing the issue that in MQTT command and previous definition we had Relay numbers starting 1 and now have it starting 0.
Will it be ok if I will change Relay numbers to start from 1?

francisjjp, 2018-05-26:
Yes, change definition in main.h so GPIO_RELAY_PIN starts from 1. I think it is easier.

soporteHW, 2018-05-28:
Does this issue performed and ready for test?

fpgamcu, 2018-05-28:
After last commit should be ready

soporteHW, 2018-05-31:
Tested. Now, relays works OK.

ralcaide, 2018-06-07:
The relay 7 does not work correctly during periodic actuation. To be fixed through #69

ralcaide, 2018-07-20:
Tested in V1.24.X

Closing the issue

closed 2018-07-20 by ralcaide