Sinapse-Energia/AP-Sinapse #142: [RF Statistics][315; - 715;]To implement the command to retrieve RF Statistics of AP

ralcaide, 2018-05-21:
To implement the commands to retrieve the RF statistics of AP

[RF Statistics](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/MQTT-Sinapse-API#get-rf-statistics)

ralcaide, 2018-05-21:
@fpgamcu , could you provide an estimation please?

fpgamcu, 2018-05-22:
Estimation is about 6 hours max, but I think I will be able to do faster.

ralcaide, 2018-05-28:
Please implement it asap. This command will help us to characterize 140;

fpgamcu, 2018-05-28:
Already done, I will run 2-3 tests and commit the code.

GonzSanch, 2018-05-31:
After this commit devices only connect with CMC context. Can you check this issue, please ?

ralcaide, 2018-06-07:
@fpgamcu , have you reproduced this problem? could please provide us feedback?

fpgamcu, 2018-06-11:
I am trying to reproduce it now. I will provide a feedback soon.

fpgamcu, 2018-06-11:
The problem is next - in 715; response we have 1%, 10%, etc. defined in wiki, but % symbol is reserved in command definition as separator, so it's usage causes HF.
I can change it to other symbol (for example 'p') in order to avoid error.

ralcaide, 2018-06-11:
Thanks. Wiki modified, p instead %

ralcaide, 2018-06-12:
We are trying to test this fix but the AP is not connecting. The gateway is stopped pointing the NTP server. 
@fpgamcu :Could you take a look? is working right in your device with the 3 different contexts?

fpgamcu, 2018-06-13:
I will check on my devices.

ralcaide, 2018-06-13:
Please, could you provide feedback over that asap? We need to perform testing and the last version of AP is not connecting.

fpgamcu, 2018-06-13:
Sorry for delay, I have tested GPRS AP and GPRS CMC - getting time from TIME server, connection to MQTT broker and answer to commands is ok.

fpgamcu, 2018-06-13:
In your case TIME server address/port could be blocked on network router/provider side.

ralcaide, 2018-06-13:
We've tested with a previous version of the code and is working right. The problem seems is not related with the TIME server but with the code. We are testing with ETH version. We will test again and I will let you know but please test also in your side with ETH version and verify the result is right. 
We are testing with the tag: AP_V1P2_F427VG_V1.20.2