Sinapse-Energia/AP-Sinapse #144: [CMC - 10;] Schedulers based in SS / SR are not working good

ralcaide, 2018-06-07:
After test the command 10; using SS / SR have been found several bad behaviors.

1.  
SR / SS = 05:07 / 19:40
10;REL11;SR+12;SR+22;-1;

a) 05:19: The relay is NOT set to ON - > KO
b) 05:30: The relay is set to OFF - > OK

Why is not set to ON when required? Maybe RTOS granularity problem?

2.
SR / SS = 05:07 / 19:40
10;REL10;SR+12;SR+22;-1;

a) 05:19: The relay is not set to OFF -> KO
b) 05:30: The relay is set to ON -> OK

3.
SR / SS = 05:07 / 19:40
10;REL11;SS+10;SS+30;-1;

a) 19:50: The relay is not set to ON -> KO
b) 20:00: The relay is set to OFF -> Why?? -> KO
c) Each 10 minutes the relay is set to OFF -> KO

ralcaide, 2018-06-07:
@fpgamcu , could you study these behaviors and reproduce the problems?   Give us also a time estimation please

fpgamcu, 2018-06-11:
Looks like granularity problem indeed, but the last one is strange.
Estimation is maximum 2 hours to reproduce (should be easy). And 2-4 hours to fix.

ralcaide, 2018-06-11:
As I commented in other task, I think will be good to encapsulate all the CMC tasks in other FreeRTOS task with RT requirements. The CMC tasks are very fast

ralcaide, 2018-06-25:
Ok, we will test also in our side

ralcaide, 2018-07-03:
The schedulers based on SS / SR are not working good. Only the schedulers based in fix hours.

Also if an on demand command is arrived (11;) the scheduler is not active anymore, so their steps should not be executed anymore, until a new 10; is send. 

@fpgamcu , could you take a look over these problems, please?

ralcaide, 2018-07-20:
@fpgamcu , please provide feedback and estimated date

fpgamcu, 2018-07-26:
There is still Bug in Context engine, changing return value in HOOKs work only for CVarstring. So in our case when you send SR / SS = 05:07 / 19:40
10;REL11;SR+12;SR+22;-1;

fpgamcu, 2018-07-26:
There is workaround, to make all time parameters not as int but as strings, should I apply it?

fpgamcu, 2018-07-26:
More description on bug:
For example in wrsouth.cpp you have next:

/*Hook on parameter AAA*/      "AAA",  NULL,   Write_AAA,
char *Write_AAA(const char *value)
{
   // If attempting to set negative value - set it to 0
   if (atoi(value) < 0) return "0";
   else return value;
}

But this code will work only if AAA is CVarstring, for CVarint it will not work.

ralcaide, 2018-07-26:
Is not possible to use strings instead int?
If not, is not possible to modify the hook (out of the engine) in order to convert date (int) to string , perform operation, and then convert again to date (int) ?

juanra-bascones, 2018-07-26:
I do not think there is a bug in the Context because of this.
What I think it happens is a misunderstanding of how SS/SR relative times have to be interpreted.

The semantic implemented in the Time-of-the-day type context variable are based in **HH:MM offsets** (no MM offsets)  (as stated in the issue https://github.com/Sinapse-Energia/M2M-CommBoard/issues/87)

So, for the case in the comment, when is 05:07 , SR+11 has to resolve to 16:12 (12 hours after Sun rise). If the goal is to set it 12 minutes after SR the command has to specify SR+00:12...

fpgamcu, 2018-07-26:
I understand, but the problem is as I explained 
When
SR / SS = 05:07 / 19:40
10;REL11;SR+12;SR+22;-1;
the exact timing for SR+12 would be set not to 05:19 but to 00:19 because of this bug, so hours value would be lost not because some problem in processing, but because HOOKs is not working as expected.
In any case I will change hours type to string and it will resolve the current problem.

Now SS/SR parameters is working like next: SS+12 means to take SS time and add 12 minutes, if need to change to SR+00:12 please tell me and I will fix it (it is a small easy fix) :)

ralcaide, 2018-07-26:
Thanks both for the comments.

@fpgamcu , if the changing of data type fix the problem without side effects, please do that. And not, is not necessary to change the SS/SR description to hours and minutes. Minutes are enough

ralcaide, 2018-08-06:
This feature is very important in the CMC and seem is still not working. 
@fpgamcu, are you able to reproduce the problem? Please provide feedback

ralcaide, 2018-09-10:
@fpgamcu ,  please provide feedback. We need this feature

fpgamcu, 2018-09-21:
Hi. I have made commit at Jul 30 that fixes the bug.
In any case no I am performing another test and will post the results of it.

soporteHW, 2018-09-21:
Hi, Sva. After commit on 30 Jul (FW V1.34), the schedulers based on SR/SS doesnt't works.
It is a strange bug, because if I test the feature "Periodic measurements" based in SR/SS time, it is working well.

So, I will repeat again the test for this issue and I will write the results obtained here.

fpgamcu, 2018-09-21:
Tested with next commands:
TX: 204;
RX: 604;342510;51;0;05:44;18:01;
TX: 10;REL41;SR+473;SR+480;-1;
At 13:37 relay 4 gone on
At 13:44 relay 4 gone off
TX: 10;REL40;SR+485;SR+487;-1;
Relay 4 sets on in less then 60 seconds (granularity for schedulers is a minute and check is performed each minute)
At 13:49 relay 4 went off
At 13:51 relay 4 went on
TX: 10;REL41;SS-246;SS-244;-1;
Relay 4 sets off in less then 60 seconds
At 13:55 relay 4 went on
At 13:57 relay 4 went off

ralcaide, 2018-09-24:
Thanks for the feedback @fpgamcu . We will perform more testing and will let you know. At first seems that with your steps works fine but if the current time is set with 127; is not working good. We will let you know the exact steps and the results

soporteHW, 2018-09-24:
Today it has been performed the following test:

1. Saved longitude and latitude params. With this params, SR = 06:15 and SS = 18:19
[image]

**2. Scheduled based on SR**
- Sended 10;REL11;SR+46;SR+51;-1; => It means switch ON at 07:00h and switch OFF at 07:05h. Results were OK in both case, although when switch ON, device Reboot!

[image]

**3. Scheduled based on SS**
- Sended 10;REL11;SS-660;SS-655;-1; => It means switch ON at 07:19h and switch OFF at 07:24h. Results were OK in both cases.
[image]

**4. RTC based on command "127"**
[image]

- **4.1 Scheduled based on SR**
 Sended 10;REL11;SR+211;SR+215;-1; => It means switch ON at 09:45h and switch OFF at 09:49h. Results were OK in both cases.
[image]

- **4.2 Scheduled based on SS**
Sended 10;REL11;SS-499;SS-484;-1; => It means switch ON at 10:00h and switch OFF at 10:05h. Results were OK in both cases.
[image]

**5. Another test with RTC based on command "127"**
[image]

- **5.1 Scheduled based on SR**
Sended 10;REL11;SR+6;SR+10;-1; => It means switch ON at 06:20h and switch OFF at 06:24h. Results were OK in both cases.
[image]

soporteHW, 2018-09-24:
So, in conclusion this issue could be mark as "Test passed".

In addition, a stability test must be performed.

ralcaide, 2018-10-05:
Test passed in V1.34

closed 2018-10-23 by soporteHW