Sinapse-Energia/AP-Sinapse #145: [CMC - Relay status] To maintain the relay status of the CMC

ralcaide, 2018-06-07:
The relay status are key in CMC product, therefore its status should be shared with the bootloader and maintained as soon as any reconnection / restart happens. The relay status should be centralized and applied as first step when the client application or the bootloader start. 

TODO:

- The relay status are in the context: Save them ALSO in the shared memory with the bootloader
- If the relay status is unknown (at the beginning), then is used OFF
- When the client application start, the relays should be set to the status saved in the shared memory
- When the bootloader application, the relays should be set to the status saved in the shared memory

Like that, the relays will be the most part of the time in the required status

ralcaide, 2018-06-07:
@fpgamcu , this is a very important issue. Please provide us feedback

fpgamcu, 2018-06-11:
For now I don't see any difficulties to implement this feature.
Estimation is 6 hours max.

ralcaide, 2018-06-11:
Ok, great. Please be aware: Should be modified the Client app AND the Bootloader

fpgamcu, 2018-06-13:
One remark:
Relay states will be saved into shared memory not at relay change but at each minute (with other device parameters), and only if it was changed. Like this we will have greatly reduced write count into shared memory.

fpgamcu, 2018-07-05:
Very strange behavior, because we have a code to put relays in needed states as soon as possible. I client even before UART and other initialization, right after GPIO init.

fpgamcu, 2018-07-11:
There was bug in relays code - absent conversion from ACTx to x and relay variables stay in 0. Now fixed.

ralcaide, 2018-07-11:
I understand the last fix is only related with client program, isn't? How are the relays when the device is in bootloader mode?. During bootloader execution, the relays should have the last saved state. 

Should I create an issue bootloader project?

fpgamcu, 2018-07-11:
It was an issue in client at saving relays into the NVM. Bootloader code is ok.

ralcaide, 2018-07-11:
Yes, I know was an issue in client. But my question is: During bootloader execution, the relays maintain the last state set in the client? It is very important for a real installation to maintain always the expected relay state

fpgamcu, 2018-07-11:
Yes, bootloader maintain the last state saved in the NVM. I have tested on my side. But please notice, that relays states is saved into the NVM with device parameters granularity (each minute).

ralcaide, 2018-07-11:
Ok, thanks.

ralcaide, 2018-07-19:
ON demmand tested passed: T367

Scheduler to be tesed: T368

ralcaide, 2018-07-23:
During tests have been observed a not desired behaviour. The lights perform 1 or 2 ON/OFF cycles when the device is restarted with 398;. It will be great to avoid this behaviour. Maybe is related with the management of the GPIOs. In other program with legacy code, this "flicker" effect is not produced.

OLD CODE:

[image]

NEW CODE:

[image]

What do you think @fpgamcu ?

soporteHW, 2018-07-24:
I performed new test with a Client FW modified as follow;

[image]

It seems that flickers disappears. Relays swtichs but the process is so fast that it is not appreciated in the lamp.

fpgamcu, 2018-07-24:
Thanks for the suggestion. Please check also with the latest pushed code, it should do the trick :)

soporteHW, 2018-07-25:
Hi, Sva. I checked your last commit and I saw that you set the gpio status before call the MX_GPIO_Initi(). 
Is this possible?

fpgamcu, 2018-07-26:
Yes, it is exactly the way ST do in GPIO initialization, first setting state of GPIO and next init it to output.

soporteHW, 2018-07-26:
OK, it's true. I saw that you changed too the GPIO_Init() function removing the RELAYS GPIO output level.
That's OK!

soporteHW, 2018-08-02:
After some tests, I can conclude that:

1) Relays don't maintain their previous status for on demmand actuation. Test performed as follow:

[image]

2) Relays maintain their previous status for scheduled actuation. Test performes as follow:

[image]

About 2) there are some notes:
2.1 Relay outputs perform two switchings after reset (398;0;)
2.2 After sending 314; command, scheduled are erased but relay outputs maintain their previous status.

ralcaide, 2018-08-06:
This feature is key in the CMC. 
@fpgamcu, are you able to reproduce the problems? We need this feature asap. Please provide feedback

ralcaide, 2018-09-10:
@fpgamcu , what is the status of this task? We need this feature

fpgamcu, 2018-10-02:
Please test the relays on restart with the latest commit, it should fix the issue

ralcaide, 2018-10-02:
Thanks, affects this modification to BL or only to the Client?

fpgamcu, 2018-10-02:
Only client. Problem was only in client.

soporteHW, 2018-10-22:
Tested passed. BUG fixed in V1.40.1

Test results here:
[Maintenance of relays status- On demand actuation](https://sinapseenergia.testrail.io/index.php?/tests/view/1102)

**NOTE:** two switchings are performed when device reboots

closed 2018-10-23 by soporteHW