Sinapse-Energia/AP-Sinapse #151: [DMX][On Demand] To implement the L3 function in order to act on demand over DMX fixtures

ralcaide, 2018-06-22:
In some projects, the RS485 interface will be used in order to control DMX lights. To implement the On Demand L3 MQTT function that will permit to act individually over DMX lights. 

Commands to implement:

[DMX on demmand](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/%5BExternal-Actuators%5D-MQTT-Sinapse-API#questions)

ralcaide, 2018-06-22:
@fpgamcu , is this task clear for you?
Regarding the code, we have already the class RS-485 that can be used in the implementation of the required DMX behaviour so I suppose the task will not be very difficult.

Could you provide an estimation, please?

fpgamcu, 2018-06-25:
The estimation is about 16 hours.

ralcaide, 2018-06-25:
Great. By the moment this task stays at backlog. I will let you know when should be done but will be soon

fpgamcu, 2018-07-04:
Info about current version of code:
DMX address parameter should be 0 to 255. DMX level 0 to 100.
DMX values saved into context, but not into NVM (set as 0 at start up).
DMX value in context stored using string, please check [parametrizacion.cpp line 928](https://github.com/Sinapse-Energia/AP-Sinapse/blob/d92d51b3dd0fbb8ebdf9a2019ebeb78f060df745/Src/parametrizacion.cpp#L928) for more details.
During to context limitation, we do not have dynamic parameters count in MQTT command, so to avoid problems you should always pass all parameters, filing with -1 not used, for example:
L3_OD_DMX;1;25;-1;-1;-1;-1;-1;-1;-1;-1;-1;-1;-1;-1;-1;-1;-1;-1;-1;-1;

ralcaide, 2018-07-05:
First version tested with result = KO:

Physically it is working but not good. Sending the same MQTT command, the Led lights make something different, changing colors and intensity.

Command sent: L3_OD_DMX;1;50;-1;-1;-1;-1;-1;-1;-1;-1;-1;-1;-1;-1;-1;-1;-1;-1;-1;-1;

fpgamcu, 2018-07-05:
I think issue could be in break generation. I am adding define that would allow to generate break at the begin of the packet (as in DMX512 standard) or not generate it.
By the way, are you able to connect USB-RS485 in parallel and sniffer packets in PC?

ralcaide, 2018-07-05:
We will try to connect a sniffer and let you know.

soporteHW, 2018-07-06:
Tested in [T495](https://sinapseenergia.testrail.io/index.php?/tests/view/495&group_by=cases:section_id&group_order=asc&group_id=24)

ralcaide, 2018-07-06:
The test finally was passed with the version 1.24.3 but this version does not contain the last changes, neither the NVM saving.
We wait until the task is totally finished and then we test again.
@fpgamcu , set the status to "To be tested" when the task is done 100% and ready to test. Thanks

fpgamcu, 2018-07-09:
> Engine: I think it is possible to have a dynamic list of arguments, so it should not be necessary to send all the unnecessary -1. Could you please test? If it is not working, it will be necessary to update the Engine.

No, unfortunately it is not possible to pass not all arguments. Because not passed arguments will be filled by context with random values.

fpgamcu, 2018-07-11:
Moving DMX values into NVM provokes HF, seems like RAM problem.

ralcaide, 2018-07-11:
Ok, but they should be saved somehow. With what context are you testing? 

Normally, when we are using DMX is not necessary to use the whole AP context or CMC context? If it is an space problem we should divide the creation of context and to decide the parts to be used with several #Define in the definitions.h

fpgamcu, 2018-07-11:
I tested with AP disabled, CMC disabled, and both disabled. I think problem is in the way we save parameters in the NVM, there is code left from the EEPROM and it is limited to the memory map. Also in current memory map implementation it is not possible to have more than one area with disable feature (like CMC). If we don't need EEPROM I can change the memory map code to SD based and fix this issue.

ralcaide, 2018-07-11:
No, we don't need EEPROM, we will use always SD. Please fix the issue. Estimation?

fpgamcu, 2018-07-11:
I will finish in 8 hours. It should also allow faster changes in future.