Sinapse-Energia/AP-Sinapse #153: [BUG][RF TX - RX] [BROADCAST / GROUPS] Acting over groups produces reconnections

ralcaide, 2018-07-04:
It has been observed a lot of AP re-connections just after acting over groups, when there are a lot of EPDs with this group

For example: Topic: ID_INST/IMEI/ACT/FFFFFA Msg: 3;60;

Why? Any idea?
- The answer bit is 0 when a message like this is sent, right?
- Maybe we need to set the transit bit also to 0?

ralcaide, 2018-07-04:
@fpgamcu , could you answer to:

The answer bit is 0 when a message like this is sent, right?

fpgamcu, 2018-07-05:
Yes, answer bit is 0.
Transmit bit is 0 in half of burst and 1 in another half. Should I change to have is 0 always for group commands?

ralcaide, 2018-07-05:
Not yet, we will perform test in our lab and I will let you know