Sinapse-Energia/AP-Sinapse #154: [BUG][Stability][1.20 .. 1.24] Devices disconnected after several weeks working

ralcaide, 2018-07-05:
9 Devices were working with V1.20 during several weeks, then updated to V1.24.2.
After 3 / 4 days, 3 devices have different  problems:

- Device a: OFF, not connected automatically. Reset HW and the device works again
- Device b: OFF, not connected automatically. Reset HW, always in bootloader. Format SD, the device works again
- Device c: OFF, not connected automatically. Reset HW, always in bootloader. Format SD, the do nothing.

Try to understand what is happening and develop stability mechanism, like reset SD option and HW reset.

ralcaide, 2018-07-05:
@fpgamcu, any idea?

fpgamcu, 2018-07-05:
Device a case: I will check the IWDG, usually it should restart the device.
Device b case: I will perform several tests with various synthetic data and FS corruptions on SD card. And implement more reliable protection.
Also I will place my device to work always online and will gather the log.