Sinapse-Energia/AP-Sinapse #159: [NVM][SD] Device should keep working even when the SD fails

ralcaide, 2018-07-11:
Now, if the SD fails or is not inserted the AP is not working by minimum basis, neither connected to the broker. 
The device should work by minimum basis even when the SD has problems or is not inserted:

EN: Commands for EN available, possibility to format the SD trough a MQTT command
CMC: Commands for CMC available. The relays should keep the expected status, including the execution of schedulers.

ralcaide, 2018-10-25:
@fpgamcu , very high priority for next sprint. What do you think about estimation?

Regarding NVM, I think will be a good idea to save everything related with the AP (CMC/EN) in flash and all the stuff related with EPD in SD. Like that, a normal CMC could work without SD at the 100% of its functionality. What do you think?

ralcaide, 2018-11-26:
@fpgamcu , what is the status of this task? . We are waiting to its development

fpgamcu, 2018-11-30:
We will push the code today evening or latest on this weekend. In any case on Monday morning you will have it.

fpgamcu, 2018-12-20:
While implementing compatibility with previous bootloader I faced issue.
Any time device is restarted bootloader is updating UPDFW_COUNT. While doing this it rewrites whole flash sector and that will erase any data outside the shared settings. So if bootloader will not be updated (old bootloader) it will always erase relays periodic settings.
We can add into shared variables some reserved field, but in any case bootloader should be updated at all devices.

soporteHW, 2019-01-02:
Is this issue ready for to be tested? @fpgamcu @ralcaide

ralcaide, 2019-01-02:
Yes, it is to be tested. In fact, Gonzalo have already tested it but we,ve found a problem. It is work only with new BL (after 1.30) and not with 1.30 and previous one. Test it please with old and new BL and characterize how is really working, then we will plan some improvements

soporteHW, 2019-01-30:
Tested using BL V1.46.0 and CL V1.46.2

[Test results - T1454](https://sinapseenergia.testrail.io/index.php?/tests/view/1454)

Performed several test in order to check basis functionality in AP device.

**- Test 1: MQTT connectivity and stability**
AP device get establishes MQTT connection and successfully subscripted to All topics.
[image]
In every main loop, AP returns messages showing SD error(as expected)
[image]
After some time, AP stability is OK. Received Ping ACK
[image]

**- Test 2: Retrieve FW version**
Sended "209;" to retrieve FW version.
[image]

**- Test 3: Set up EPD**
Two EPDs set up with ID 514853, FFFFFF 
[image]

**- Test 4: Discover EPD in memory**
Sended "313;" to discover the EPD stored in RAM & Flash memory
[image]

**- Test 5: Reset the AP**
Sended "398;0;"
[image]
AP reboots as expected and establishes again the MQTT connection.
Then, when we discover the EPD in memory, the memory is empty.
[image]

**- Test 6: RF communication: Actuation over EPD**
With EPD set up, sended "3;X;". EPDs work as expected (switch ON/OFF, dimming)

**- Test 7: RF communication: metering over EPD**
Sended "1;" over EPD 514853
[image]
All values returned are 0. It is due a CL V1.46.2 RF NEMA improve. In this version, the communication with EVO EPD is worst.

**- Test 8: Actuation over Relay outputs(On demand)**
Sended some actuations as follow:
[image]
In all cases, relays switch as expected and AP device returns "L3_READ_RELAY_STATUS_R" message with last relay status.

**- Test 9: Actuation over Relay outputs(Scheduled)**
Performed a scheduled actuation over REL1, REL2 and REL3.
[image]
When scheduled is enable, if we send a "L3_READ_RELAY_STATUS_OD;" message, AP device shows the actual relay status after change.
Idem when scheduled is disabled.
[image]

**- Test 10: Metering over MODBUS analyzer**
Sended "L3_READ_METERING_OD;" message:
[image]
Metering works as expected.

**NOTE:** Periodic features are not tested.

ralcaide, 2019-02-19:
Working with BL 1.46 or higher.
It is not working with BL 1.30. It is necessary to find a workaround maybe with limitations but that allows the device to be update with last CLs

ralcaide, 2019-02-19:
@fpgamcu , could you please provide options we have?

closed 2019-03-28 by ralcaide