Sinapse-Energia/AP-Sinapse #160: [AQP][L3 - On demmand] Develop L3 commands to read Air Quality sensor in demmand

ralcaide, 2018-07-17:
To develop the MQTT command to read Air Quality sensor(s) on demmand:

[L3 AQP On demmand](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/%5BExternal-Sensors%5D-MQTT-Sinapse-API#air-quality--particles)

Based on [Generic UART](https://github.com/Sinapse-Energia/AP-Sinapse/issues/107)

ralcaide, 2018-07-17:
@fpgamcu , could you give us an estimation about this task? Is everything clear?

fpgamcu, 2018-07-23:
5 hours should be enough. Do you have datasheet on the sensors that would be used?

ralcaide, 2018-08-01:
@fpgamcu, this command have a high priority right now. Please focus on this one at the beginning of the current sprint.

When could we have a first testable version?

fpgamcu, 2018-08-01:
Sure, now working on it. Thursday evening I will push the code.

ralcaide, 2018-08-01:
Ok, so tomorrow evening, we will plan first test on Friday. Thank you

ralcaide, 2018-08-03:
Great, we will provide you with an external with the sensor

fpgamcu, 2018-08-03:
To be 100% sure, On Demand is performed like next:
1) Triggering the measurement
2) Wait 6s for result
3) Send result over MQTT

Is it ok, or the operation should be performed in different way?

ralcaide, 2018-08-03:
Yes, but the values should be saved in NVM with timestamp. Next time, if the sensor does not answer should send last measurement

fpgamcu, 2018-08-03:
Ok, several more questions:
1) If we specify for example M6, should other parameters remain as was, or changed to X? For example what is true:
a) L3_OD_AQP;M6; -> L3_OD_AQP_R;123456;X;X;X;X;X;3;X;1234;
b) L3_OD_AQP;M6; -> L3_OD_AQP_R;123456;X;X;X;X;1;3;2;1234;
2) If we specify ALL, should timestamp be changed if any parameters was updated, or if all was updated:
what is true:
a) L3_OD_AQP;ALL; -> (TS was 1234, answered only M2) -> L3_OD_AQP_R;123456;X;2;X;X;1;3;4;1235;
b) L3_OD_AQP;ALL; -> (TS was 1234, answered only M2) -> L3_OD_AQP_R;123456;X;2;X;X;1;3;4;1234;

ralcaide, 2018-08-03:
1:a), 2: a)

fpgamcu, 2018-08-03:
ok, so in L3_OD_AQP_R; not needed parameters should be changed to X in response, but remain as was in NVM.

fpgamcu, 2018-08-03:
also now there is only one timestamp for all parameters, so we could have situation like next:
0) TS was 1230
1) L3_OD_AQP;M6;
2) got correct answer over uart
3) L3_OD_AQP_R;123456;X;X;X;X;X;3;X;1234;
4) L3_OD_AQP;M7;
2) got incorrect answer over uart or no answer
3) L3_OD_AQP_R;123456;X;X;X;X;X;X;X;1234;
is it ok?

ralcaide, 2018-08-03:
It's strange but we leave like that by the moment, let me think

ralcaide, 2018-08-14:
Tested OK in V1.30.2