Sinapse-Energia/AP-Sinapse #165: [BUG][Bootloader configuration][207; - 208; - 607;] Bootloader configuration commands are not working as expected

ralcaide, 2018-07-26:
This task is related with #119 and #134.

The command 607; is not working as expected after a 207; and 208;. 

The 607; should contains always the current values that will be used by the BL. These values are explained here:

[BL configuration](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/MQTT-Sinapse-API#bootloader-configuration-information)

- To add FW_NAME: New added
- Response to 208; should be a valid 607. Now is:

[image]

- Response to 207; should be valid. For example, when only the UPDATE_FW is update we have the following response

[image]

Problem detected in 1.24

soporteHW, 2018-07-26:
Tests performed here:

207) https://sinapseenergia.testrail.io/index.php?/tests/view/540
208) https://sinapseenergia.testrail.io/index.php?/tests/view/552

ralcaide, 2018-08-14:
@fpgamcu , status of this task? Please provide feedback: reproducible? estimation?

ralcaide, 2018-08-21:
Please, provide feedback

fpgamcu, 2018-08-28:
Question about 607; response: I noticed that if some value is empty string (for example PATH) engine is not adding this parameter into response, so instead of `607;012345;0;1.30.3;www.dummy.org;80;HTTP;;a.bin;;;3;` we receiving `607;012345;0;1.30.3;www.dummy.org;80;HTTP;a.bin;3;`. Should I fix this behavior?

ralcaide, 2018-08-28:
Yes please, the right option is the first one

soporteHW, 2018-09-05:
Testing results OK. Test passed.

closed 2018-10-17 by soporteHW