Sinapse-Energia/AP-Sinapse #166: [BL-Client][V1.30] The bootloader - client integration is not working well with different contexts

ralcaide, 2018-07-31:
After start testing, seems that the only context that is working is the AP. See the following table

[image]

Try to reproduce and fix

ralcaide, 2018-08-06:
Testing with V1.31.1

- EN GPRS: KO. The SD is empty after download the client and trying to connect
- AP GPRS: OK.

@fpgamcu , have you tested with context EN?. In our test is not working

soporteHW, 2018-08-07:
Table updated at 07/08/2018:

[image]

Testing with V1.30.1

fpgamcu, 2018-08-08:
Please check also with latest commit. It should fix the EN bug.

fpgamcu, 2018-08-08:
Also I have added definitions DEVICE_CMC, DEVICE_AP and DEVICE_EN to more easily select needed context.

soporteHW, 2018-08-09:
Last test result for FW V1.30.2 was:

[image]

All contexts connects OK

ralcaide, 2018-08-14:
Tested OK