Sinapse-Energia/AP-Sinapse #167: [NB - Scheduler - SB] Unplug NB connection (MQTT) from the rest of scheduled operations

ralcaide, 2018-08-08:
The device should perform every (possible) scheduled action when the internet connection is not available. It is to say, the connection to internet should be unplugged from the rest of the features.

For example, if the internet connection is not available, the device should still perform the scheduled relay actuation, scheduled RF actuation, scheduled RF question (without publishing results), scheduled SB measurements like Modbus (without publishing results), etc.

Critical grade:

High: Relay actuations in CMC
Medium: RF actuations
Low: Metering questions (RF, Modbus, etc)

ralcaide, 2018-08-08:
@fpgamcu , we need asap the high critical part of this task (relay actuation). Could you provide please feedback and estimation by critical grades?

ralcaide, 2018-08-09:
After some tests, we see that is not working as expected. Also when some scheduled tasks are set and the connection is lost, the SD is corrupted and the device stop working. 

@fpgamcu , we need to have a device consistent when the internet connection is lost. It should try to connect always and restart after N min if the connection is not performed but in the meanwhile should continue with its operations. 

This is KEY for our projects. Could you please provide feedback?

fpgamcu, 2018-08-09:
Hi. I am working now on this feature. I expect to have it done tomorrow, maximum at the end of the day, but should be faster, I think even at the middle of the day. The issue now is that reconnect process is blocking the engine mutexes, but I will fix it.

ralcaide, 2018-08-09:
Thank you very much

ralcaide, 2018-08-14:
Hello @fpgamcu , what is the status of this task? As you know, this feature is very important for our product

ralcaide, 2018-08-21:
Hello @fpgamcu , could you please provide feedback and estimation. This task is blocking some of our projects

fpgamcu, 2018-08-23:
I have tested it in the next way: Context AP, GPRS.
Wait for connection, send 11;ACTX;ACTX;ACTX;ACTX;ACTX;ACTX;ACT1; and ensure relay7 is high.
Send 10;REL70;13:45;14:00;-1; to set relay off in 13:45 and on again in 14:00.
Removed SIM and restarted device.
In 13:45 relay switched to 0. in 14:00 - switched to 1.

ralcaide, 2018-09-10:
Tested OK: https://sinapseenergia.testrail.io/index.php?/tests/view/1031