Sinapse-Energia/AP-Sinapse #168: [Automatic Builds] Build with GNU Make Builder fails

ralcaide, 2018-08-09:
We are tying to automatize the builds generation and as a first step we want to use the GNU Make Builder instead of the CDT Internal Builder.
Problem found: With the GNU Make builder we are not able to compile, it fails.

Is necessary to find the reason and provide a solution

ralcaide, 2018-08-14:
@fpgamcu , are you able to compile using the GNU Make Builder? If not, any idea about the reason?

ralcaide, 2018-10-05:
@fpgamcu , we are investing a lot of time in the build process, this is why we want to automate it. Could you help us with that, please? Provide feedback and we will see how to proceed

ralcaide, 2018-10-15:
Could you provide an example about how to create a .bin?

I understand that now is possible to compile using make but is possible to change context variables or comment / un-comment defines from Definitions.h? I understand not, right? That should be planned in other task

ralcaide, 2018-10-29:
@fpgamcu , could you please answer my previous questions?

fpgamcu, 2018-10-30:
Usage: from "build" directory
make all - to make current code and prepare bin and hex
make cmc - switch definitions to CMC and prepare bin and hex
make ap - same but for the AP
make en - same but for the EN

ralcaide, 2019-03-12:
Testing...
During tests we've found several improvements required in order to do not modify the code. Now it is possible to choose between contexts but this is not enough to automatize the whole process. We need also:

- GPRS or WIFI/ETH
- If GPRS: APN, USR, PASWD
- If WIFI/ETH: DHCP , STATIC (FIRST_TIME_CONFIG = 1) , WIFI (FIRST_TIME_CONFIG = 0)
- Broker, user, password
- Topic
-  CUR_FW_Version

@fpgamcu , what do you think? is feasible? Estimation, please.

fpgamcu, 2019-03-18:
Yes, it can be done through make file
Estimation is 16 hours

ralcaide, 2019-03-22:
The makefile should setup also a the new variable BINARY_INFO. See #206

fpgamcu, 2019-03-27:
Example config and make commands flow:
```make clean``` to clean compiled binaries
```make change_cmc``` or ```make change_ap``` or ```make change_en``` to select context
```make change_gprs``` or ```make change_wlan``` to select communication
```make config_gprs APN=internet USR=user PASSWD=1111``` to config GPRS
```make config_gprs APN=internet USR= PASSWD=``` to config GPRS with empty user and pass
```make config_wlan_mode_dhcp``` to switch WLAN to DHCP
```make config_wlan_mode_static``` to switch WLAN to static
```make config_wlan_ftc_enable``` to enable FIRST_TIME_CONFIG
```make config_wlan_ftc_disable``` to disable FIRST_TIME_CONFIG
```make config_wlan_static STIP=192.168.5.1 STMASK=255.255.0.0 STGATEWAY=192.168.5.1``` to config static WLAN parameters
```make config_mqtt_broker MQTTURI=test.com MQTTPORT=1234 MQTTUSER= MQTTPASS=``` to config MQTT broker
```make config_mqtt_topic MQTTTOPIC=TESTX"``` to config topic
```make config_version FWVER=1.46.3``` to config FW version
```make config_info BINFO='Test in lab'``` to config binary info
then
```make all``` to build binary

ralcaide, 2019-03-28:
Thank you very much. We will test and provide you feedback

ralcaide, 2019-04-12:
@fpgamcu , in Linux is not working properly:

Compilling ../Src/tda5340.c
../Src/tda5340.c:22:21: fatal error: TDA5340.h: No such file or directory
 #include "TDA5340.h"
                     ^
Is it possible to fix this problem with a compilation option in the Makefile?

ralcaide, 2019-04-12:
Preliminary tests are not working good. Strange thing:
- binary created using makefile: 
-rwxr-xr-x 1 1000 1000  209752 Apr 12 13:44 AP-Sinapse.bin
- binary created from Atollic:
-rwxr-xr-x 1 1000 1000  209740 Apr 12 13:45  AP-Sinapse.binary

There is 12 bytes difference. Why? Are you adding the CRC directly?

ralcaide, 2019-04-12:
Tested:

AP GPRS : KO
EN ETH : OK

@fpgamcu, could you please test if AP GPRS works in your device? Version: 191200. See tags

fpgamcu, 2019-04-12:
Better would be to fix tda5340.c to include TDA5340.H instead of TDA5340.h
Binary should be the same, I'll check
Ok, will test GPRS. Need to test with bootloader?

ralcaide, 2019-04-12:
ok. We've modified the tda code in local.

AP GPRS: Yes, we are using BL 1.46.0
EN ETH: We are using also BL 1.46.0 and works OK

We are using the makefile to create the binaries

`#Global variables
APN=matooma.m2m
VERSION=191200

###############################################################################################
# AP 
TOPIC=AP_GENERIC

###################################################
# AP GPRS
INFO='AP_GENERIC_GPRS'

make clean
make change_ap
make change_gprs
make config_gprs APN=$APN USR= PASSWD=
make config_mqtt_broker MQTTURI=soporte-tecnico.bitnamiapp.com MQTTPORT=1883 MQTTUSER= MQTTPASS=
make config_mqtt_topic MQTTTOPIC=$TOPIC
make config_version FWVER=$VERSION
make config_info BINFO=$INFO  
make all

cp $ORIG_BIN $DEST_PATH$DEST_BIN
CRC=$(crc32 $DEST_PATH$DEST_BIN)
echo CRC32 is $CRC
perl -e 'print pack "H*", "'$CRC'"' >> $DEST_PATH$DEST_BIN

mv $DEST_PATH$DEST_BIN $DEST_PATH$VERSION'/CL-'$INFO'-'$VERSION'.bin'`