Sinapse-Energia/AP-Sinapse #169: [Schedulers][AP] Scheduled actions over CMC & EPD in parallel break the SD and the system

ralcaide, 2018-08-09:
Context AP
Scheduled actions over relays
Scheduled actions over EPDs (for example 4;)

The device restarts and stay locked in Bootloader

ralcaide, 2018-08-14:
@fpgamcu , please give feedback. Is reproducible in your side? Estimation to have the fix?

ralcaide, 2018-08-21:
@fpgamcu , could you please provide feedback and estimation?

ralcaide, 2018-09-04:
Thanks for the feedback. 
Sorry, but there is a mistake in the task explanation. The problem does not appears with the command 4; over the EPDs but with the command 2;. 
Also, the problem with the command 2; appears also without schedulers over relays.

The problem is: The device is restarted and the SD is corrupted. The device is blocked in the Bootloader.

@fpgamcu , please take a look and provide feedback

Attached files found in the SD after the restart 
[SD_files.zip](https://github.com/Sinapse-Energia/AP-Sinapse/files/2348286/SD_files.zip)

ralcaide, 2018-09-10:
After testing V1.34.0:

It seems the problem is only related with the command 2; over an EPD. Here is the result after act with 2; over just one EPD but in the Context there are 8 EPDs plus the device representing broadcast FFFFFF:

Actuation based on scheduled time

Perform a scheduled actuation over some EPDs using the command '2;DIM1;HH1:MM1;DIM2;HH2:MM2;'

Expected Result
The EPDs switch as expected

Actual Result
Sended the following schedule actuation over the EPD 514853 with the result indicated:

1) 2;0;09:00; => OK
2) 2;50;09:02; => NOK
3) 2;50;09:10; => OK
4) 2;100;09:15; => NOK
5) 2;0;09:20; => OK
6) 2;100;09:25; => NOK

Then, in order to delete the scheduler, it has been sended the command "2;-1;00:00;" => DUT get in BL proccess and then it stays locked.

After, erased the SD card, a new scheduler has been saved as "2;0;09:45;100;09:50;" => DUT get in BL process and then it stays locked.

After, erased the SD card, a new scheduler has been saved as "2;0;10:00;100;10:10;" => DUT get in BL process and then it stays locked.

Attached the SD files saved in test failed.

[SD_files_2.rar.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/2365844/SD_files_2.rar.txt)

@fpgamcu , could you please try to reproduce the problem a provide a solution.

ralcaide, 2018-09-10:
Could this problem be related with: https://github.com/Sinapse-Energia/AP-Sinapse/issues/158 ?

Maybe is a good idea to integrate the fix

ralcaide, 2018-10-05:
@fpgamcu , please provide feedback about that. It is very important to have a system as stable as possible and corrupt the SD is a big problem

fpgamcu, 2018-10-15:
If you remember several times ago we changed the scheduler to have granularity 10 minutes, so next schedules will not work:

fpgamcu, 2018-10-15:
Also as we discussed before, our Engine don't have such feature as dynamic parameter count for input message, so when you publish any message with lesser parameters number than needed, you will get unpredictable behavior. Some random memory regions could be changed. Some random code parts could be executed. It is leading to possible erasing or corrupting external memory, or shared with bootloader memory.

fpgamcu, 2018-10-15:
When I tried to execute most of the next commands, I got HF most of the time:
2;0;09:00;
2;50;09:02;
2;50;09:10;
2;100;09:15;
2;0;09:20;
2;100;09:25;
2;-1;00:00;
2;0;09:45;100;09:50;
2;0;10:00;100;10:10;

ralcaide, 2018-10-15:
Ok, thank you for the remember. We will test again. 
Anyway, have been integrated the fix proposed in #158 ? Maybe after this fix is allowed dynamic number of  parameters?

fpgamcu, 2018-10-15:
I will integrate and test #158 asap. Seems like it will not totally fix this danger behavior. But I think I can debug step by step and add a check in needed place to process this

soporteHW, 2018-10-17:
Some tests performed fot this issue:

**1) Paralell test:**
- Periodic message over EPD
- Scheduled actuation over Relay output
- Scheduled actuation over EPD

[image]

All the actions works as expected: periodic message received, relay output switches and EPD acts.

ralcaide, 2018-10-19:
To create a bug about tcp connection

ralcaide, 2019-03-12:
To manage it