Sinapse-Energia/AP-Sinapse #171: [CMC - 10;][BUG] To improve time interval management

ralcaide, 2018-08-20:
The actuation with time of different days are not working. For example this one: 
`10;REL11;20:48;07:03;-1;`

The code developed in #69 should be improved with the same considerations explained in #125

ralcaide, 2018-08-20:
@fpgamcu , this bug is creating big troubles in real installations. Could you please give us an estimation and work on it with high priority?

ralcaide, 2018-08-20:
By the moment we have found a workaround that is working in our projects:

10;REL11;20:48;07:03;-1; == 10;REL10;07:03;20:48;-1;

The priority is medium and the issue will be at the backlog until finish high priority tasks

ralcaide, 2018-10-19:
@fpgamcu , could you give us estimation?. Same as #125

fpgamcu, 2018-10-19:
6 hours should be enough

soporteHW, 2018-10-23:
**NOTE**: #125 fix this problem for EPD devices. It's necessary to check this behaviour in "L3_READ_METERING_P" command.