Sinapse-Energia/AP-Sinapse #172: [BUG][API 5;] The connection to other broker is not well managed

ralcaide, 2018-09-10:
After some testing, the reconnection to other broker seems that work fine but after restart the device with 398;0; the device connects again to the original broker.

The device connects always to the contingency broker

[image]

fpgamcu, 2018-10-15:
Was bug here - displaying main broker address while connecting to backup broker.
Now when connect to backup broker will display backup broker address.

soporteHW, 2018-10-16:
Tested with FW V1.40.1 => Test OK

Results here: [API 5; Change MQTT Broker credentials](https://sinapseenergia.testrail.io/index.php?/tests/view/1108)

closed 2018-10-16 by ralcaide