Sinapse-Energia/AP-Sinapse #173: [BROADCAST] The APs (CMCs & EN) should listen also in the Broadcast address and process the messages received

ralcaide, 2018-09-14:
Now, each AP has an ID (IMEI or MAC) and is listening there. It is necessary to set up also a broadcast address with ID FFFFFF, like that will be possible to act over N APs at the same time with only one command or for example to act over one EPD through N APs (if it is working in transparent mode).

Example:

ID_INST/FFFFFF/ACT/100101 3;100;

Will the a SET_ON to the device 100101 through all the APs with ID_INST

fpgamcu, 2018-10-02:
Last commit should unlock possibility to use group AP id in EPD related commands. In other commands this feature was already implemented.

ralcaide, 2018-10-05:
Thanks, will be tested

soporteHW, 2018-10-16:
Tested in FW V1.40.1 => Test OK

Test results here: [Broadcast Features](https://sinapseenergia.testrail.io/index.php?/tests/view/1332)

closed 2018-10-17 by soporteHW