Sinapse-Energia/AP-Sinapse #174: [MQTT Processing][Security] Avoid message processing received trough not valid topics or not valid message

ralcaide, 2018-09-19:
At this moment, the MQTT Client of the device is processing almost anything that is receiving and this is very dangerous. For example:

**Message received in wrong topic is processed**

TOPIC: AP_GENERIC_V1P2/072011/ACT/514853 (EPD TOPIC)
MESSAGE:  11;ACT1;ACT1;...

The topic is related with an EPD and the message with a CMC. The message is processed and should not

The topic AP_GENERIC_V1P2/072011/ACT/XXXXXX is also working. This is very dangerous.

**Not valid message received through valid topic broke the device**

TOPIC: AP_GENERIC_V1P2/072011/ACT/514853 (EPD TOPIC)
MESSAGE:  false

The SD is corrupted, the device restart and then does not start again. This is a **huge** problem

The device should process only valid messages in valid topics

ralcaide, 2018-10-03:
Question: How we are getting now the EPD_ID from the topic?

ralcaide, 2018-10-08:
Ok, we will study it.

Anyway, the other problem is not related with the topic:

send "false" as a message to a valid topic breaks the SD. @fpgamcu , are you able to reproduce that?

fpgamcu, 2018-10-09:
I will debug it today.