Sinapse-Energia/AP-Sinapse #175: [AQP][UART] Avoid hang up / block of PM particles sensor communicating through UART

ralcaide, 2018-09-20:
Issue related with: #107 and #160 
At this moment, both commands are working good but only during several hours, then the sensor is blocked, the AP continues working normally but the answer of the sensor is always the same. 

To understand the problem and fix the behaviour.

Sensor: https://www.mouser.es/ProductDetail/Honeywell/HPMA115S0-XXX?qs=sGAEpiMZZMtWSrBd5SaE4GJSel3MVzU7Fep%252b%252b8YyEDhkpP27cz%2feGQ%3d%3d

fpgamcu, 2018-10-09:
Just yesterday evening received sensor. Today I will debug its operation.

ralcaide, 2018-10-09:
Great, please provide feedback as soon as you know something more. Our sensors are working between 2-4 hours, then should be physically restarted and this is not acceptable

fpgamcu, 2018-10-11:
I have performed 12 hour test with sensor we received. With latest commit hang up is not appeared.
But as I remember sensor that you have starting to auto send data when you connect it. This sensor I have is not auto sending data, more it is not reacting to stop auto send command and this cause measurement operation to stop because of no reaction. Latest commit should fix such kind of situations, even if sensor is not answering to stop autosend command, further commands would be transmitted.

ralcaide, 2018-10-11:
Thanks, before this commit, were you able to reproduce the problem?

ralcaide, 2018-10-29:
@soporteHW , please update with the test results