Sinapse-Energia/AP-Sinapse #176: [M95] [USR] Management of the reconnection / RESTART

ralcaide, 2018-09-24:
How is working now reconnection to MQTT Broker when connection is lost? Could you explain the procedure, please?

**Regarding M95**

Is the Quectel somehow physically restarted? In the current AP board is possible to restart the M95. Maybe it is a good idea in order to avoid device hang up

**Regarding USR**

Is the USR somehow restarted when is not able to connect? Maybe is a good idea in order to avoid device hang up

ralcaide, 2018-09-24:
@fpgamcu , could you at least provide feedback about how is the procedure when the broker connection is lost?

fpgamcu, 2018-10-02:
M95 - no, it is not physically restarted. But, working on TF project, I noticed that physical restart is much better than only reconnection by commands.
Would be good to modify FW to send AT+QPOWD=1 on disconnect to power off the M95 and than power it on with pins.

fpgamcu, 2018-10-02:
USR - yes, it is restarted at each reconnection.

ralcaide, 2018-10-08:
We need more info about the connection / reconnection procedure when working in with M95. We have a device consuming more than 200KB each 3 minutes and we don't know how is that possible. Also is creating big problems for us.

Questions:

- How many times can the device try to reconnect in 1 to 60 min? Can we limitate that somehow?
- Any idea about this huge consumption? debug messages?

fpgamcu, 2018-10-09:
Now unlimited reconnections. We can add a limiter.

On consumption problem need to investigate more. We have this debug messages, but it should not consume 200KB in 3 min, from other side 200KB it is traffic in the both sides, and we also receive a messages from broker. Also how you measured 200KB, it is got from mobile provider? Is it exactly 200KB? Because it is very round number, not 215310 bytes or so, but 200KB, this means that mobile provider have so called traffic rounding. For example by provider rounds traffic to 1MB, so if I open connection, send or receive something, and close connection, it will round my traffic to 1MB, even if I transmit 1 byte. Also need to get info from provider, this 200KB is exact data bytes, TCP bytes or channel level bytes? Because TCP header takes some bytes, and channel level also takes some bytes for its header.

ralcaide, 2018-10-09:
- Yes: We will add a limit, I will create the task for that. We would like to have something like 5 times max in one minute and 30 times max per hour. Is that possible?. These numbers should be dynamic
- Consumption of 200KB in 3 min: The number comes from a provider but is not a round number. Is something like that:

2018-09-04 | 18:33:13 |   |   | MATOOMA.M2M | 0,2002 | DATA (MB) | Orange
-- | -- | -- | -- | -- | -- | -- | --
2018-09-04 | 18:36:13 |   |   | MATOOMA.M2M | 0,19922 | DATA (MB) | Orange
2018-09-04 | 18:39:06 |   |   | MATOOMA.M2M | 0,2002 | DATA (MB) | Orange
2018-09-04 | 18:42:07 |   |   | MATOOMA.M2M | 0,19824 | DATA (MB) | Orange
2018-09-04 | 18:45:01 |   |   | MATOOMA.M2M | 0,19824 | DATA (MB) | Orange
2018-09-04 | 18:47:58 |   |   | MATOOMA.M2M | 0,20117 | DATA (MB) | Orange
2018-09-04 | 18:51:10 |   |   | MATOOMA.M2M | 0,20117 | DATA (MB) | Orange
2018-09-04 | 18:54:11 |   |   | MATOOMA.M2M | 0,2002 | DATA (MB) | Orange
2018-09-04 | 18:57:14 |   |   | MATOOMA.M2M | 0,19922 | DATA (MB) | Orange

We have now more information about the what happens the days of this big consumption: The SD card holder was broke, so normally was not possible the communication with the SD.

Could this fact be related with? Anyway, we should design a mechanism to limitate the consumption. 200KBs each 3 minutes could be a huge problem if that appears in several devices