Sinapse-Energia/AP-Sinapse #177: [AQP][L3 - Periodic][PM]Develop L3 commands to read Air Quality sensor periodically

ralcaide, 2018-10-04:
To develop the command

[AQP Periodic](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/%5BExternal-Sensors%5D-MQTT-Sinapse-API#periodic-2)

Related with [AQP Demmand](https://github.com/Sinapse-Energia/AP-Sinapse/issues/160)

The period by default is 0 but should be modifiable in Context/Definitions in order to set a different period by default, for example 10 minutes.

ralcaide, 2018-10-04:
@fpgamcu , I think this development is very fast, could you please do it asap? estimation?

fpgamcu, 2018-10-08:
Default period is customizable through Definitions.h L3_DEFAULT_PERIOD definition.
In implementation period is in minutes (not in seconds as described in wiki) since we have scheduler granularity 1 minute.
Please also notice that according to wiki publication topic is ID_INST/CMC/MEASUREMENTS, not ID_INST/CMC/PERIODIC

ralcaide, 2018-10-08:
Thanks.
I've modified the wiki:
- Period in minutes
- Topic for periodic should be the PERIODIC period instead the MEASUREMENTS period

ralcaide, 2018-10-08:
Please change the topic and then set again to task to "To be tested"

fpgamcu, 2018-10-08:
There is a mistake here. Link in this task [AQP Periodic](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/%5BExternal-Sensors%5D-MQTT-Sinapse-API#periodic-2) is related to T and H periodic so here exactly implemented T and H periodic. I will fox the topic and add AQP periodic asap

ralcaide, 2018-10-08:
Sorry, I've updated the link. This task is related with Air Quality and not T&H

soporteHW, 2018-10-09:
Tested OK

Test results here:
[AQP - Periodically](https://sinapseenergia.testrail.io/index.php?/tests/view/1303)

closed 2018-10-17 by soporteHW