Sinapse-Energia/AP-Sinapse #181: [CMC][Relay status] Implement API commands to get the relay status

ralcaide, 2018-10-25:
To implement the following commands:

[Ask relay status - on demand](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/MQTT-Sinapse-API#ask-relay-status-on-demand)

[Ask relay status - periodically](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/MQTT-Sinapse-API#ask-relay-status-periodically)

soporteHW, 2018-11-15:
Testing **_Ask relay status - On demand_**

**- Case 1:** switching ON all relays outputs

[image]

Received "L3_READ_RELAY_STATUS_R;" in MEASUREMENTS topic when command "11;" is sended.
After, when "L3_READ_RELAY_STATUS_OD;" is sended, as image shows in message response, relays status value are 1 as expected, but TIMESTAMP value is the same all the time. This TIMESTAMP is the last value returned when command "11;" is performed.

**- Case 2:** changing relays status

[image]

Again, received "L3_READ_RELAY_STATUS_R;" in MEASUREMENTS topic when command "11;" is sended.
After, when "L3_READ_RELAY_STATUS_OD;" is sended, as image shows in message response, relays status value are 1 as expected, but TIMESTAMP value is the same all the time. This TIMESTAMP is the last value returned when command "11;" is performed.

**-Case 3:**  scheduled actuation performed over REL1

[image]

Received "L3_READ_RELAY_STATUS_R;" in PERIODIC ¿? topic when command "11;" is sended.
Now, as image shows in message response, relay status value are the same as previous status, but TIMESTAMP value change.

At 15:45h when scheduled actuation is activated, if "L3_READ_RELAY_STATUS_OD;" is sended, the message responses are:

[image]

REL1 status value change 1 to 0 as expected, and TIMESTAMP change over time.

**- Case 4:** testing X feature

4.1 REL1 = 1; REL2 = 1; REL3 = 1; REL4 = 1; REL5 = X; REL6 = X;

[image]

As image shows, REL1, REL2, REL3 and REL4 change their previous status to 1, REL5 keep 0 value but REL6 change status 1 to 0 (unexpected result.  API command definition writes => X=Not present).
And now, TIMESTAMP doesn't change over time.

4.2 All REL = X

[image]

In this case, all relays status change to 0 (unexpected result). Also, TIMESTAMP doesn't change over time.

soporteHW, 2018-11-15:
Testing **_Ask relay status - Periodically_**

**- Test 1:**
[image]

Received 10 x "L3_READ_RELAY_STATUS_R" as expected. Bug: all message with same TIMESTAMP value.

**- Test 2:** Scheduled based in differents days.
[image]

Only 1 x "L3_READ_RELAY_STATUS_R;" message received after the request command.

ralcaide, 2018-11-16:
@soporteHW , for me results are the following

On demmand:

- TEST1 - Result is OK: the timestamp is the value of the last modification of the relays
- TEST2 : Same as TEST1
- TEST3 -  Result is KO: the timestamp should change only when a relay status is modified
- TEST4: Yes, result is KO, the when X is sent, the relay should stays in its previous status. Question: The real relay maintain the status? Is only a message problem or a relay problem?

Periodically:

- TEST1 - Result is OK: the timestamp should be modified only when the relay status change
- TEST2 - Result is KO. To be fixed

ralcaide, 2018-11-26:
@fpgamcu , could you please fix the little bugs found during the testing:
- Timestamp should change only when a Relay status is modified
- When a relay is set to X, the real status and the status in the NVM should not be modified

fpgamcu, 2018-12-03:
>Timestamp should change only when a Relay status is modified
If I understand right, when relay is already for example in '1' and you set it in '1' with on demand or periodic, timestamp should not update. Is it correct?

soporteHW, 2019-01-29:
**CONCLUSIONS** about _Testing **Ask relay status - On demand**_

1) On demand actuations over relays: L3_READ_RELAY_STATUS works as expected. Messages returns the last relay change status.
2) Actuations based on scheduled time: L3_READ_RELAY_STATUS returns the relay status in real time, not the last change status.

soporteHW, 2019-01-29:
Testing **_Ask relay status - Periodically_**
FW: BL V1.46.0 and CL V1.46.2

[Test results - T1417](https://sinapseenergia.testrail.io/index.php?/tests/view/1417)

**Case 1:** L3_READ_RELAY_STATUS_P based on fixed time (short time of 10 minutes)
[image]

**Case 2:** L3_READ_RELAY_STATUS_P based on fixed time (long time of 60 minutes)
[image]

**Case 3:** L3_READ_RELAY_STATUS_P based on fixed time (different days)
[image]

**Case 4:** L3_READ_RELAY_STATUS_P based on fixed time (based on SR/SS)
[image]

**CONCLUSIONS:** DUT works as expected except if the periodicity covers different days

ralcaide, 2019-01-31:
Thanks @soporteHW for the great analysis. Resuming, there are two problems in the periodic publications of the relay status:
- The status is published every minute but should be published only when the status of a relay changes.
- The periodic publications don't work if the hours are from different days.

@fpgamcu , could you please take a look to these problems and provide us an estimation about the fixings?

fpgamcu, 2019-02-06:
Now fixing the issue, should take about one day maximum

ralcaide, 2019-02-07:
@fpgamcu , could you update the status?

ralcaide, 2019-02-11:
@fpgamcu , analyzing some of our costumers requirements we found we lack a command to get the schedulers of the CMC relays. It will be necessary to get what programs are set to each relay or if it is acting on demand. 
What do you think about to add this information to the here explained command? Is it viable? Or do you think is better to add this information into a new command?

I imagine something like:

1. L3_READ_RELAY_STATUS;ID;R1_STATUS,R1_SCHEDULE;...;R7_STATUS;R7_SCHEDULE;TIMESTAMP; 
Where:

RX_SCHEDULE can be: 

OD : On demmand

OR

0,SR-10,SS+10 : Indicates the interval where the relay is 0

OR

1,SS-10,23:30; : Indicates the interval where the relay is 1

What do you think?

fpgamcu, 2019-02-14:
I think it is good idea and easy to do, will add it now and commit along with the periodic fixes
one question, maybe also include NTIMES to RX_SCHEDULE, like 0,SR-10,SS+10,-1

ralcaide, 2019-02-14:
Yes please, include all the info related with the schedule

ralcaide, 2019-03-08:
Hi @fpgamcu , could you please provide feedback about that?. Is it already done?
In other order, is it possible to send the evaluated hour regarding the SR and SS time in relation with the coordinates they have?

ralcaide, 2019-03-11:
Issue found:  SS / SR is calculated only when the 10; is sent. It should be calculated very day

ralcaide, 2019-03-12:
Moved to new Milestone 12-19

ralcaide, 2019-04-12:
@fpgamcu , status? Is it possible to finish the milestone 19.12 something soon?