Sinapse-Energia/AP-Sinapse #183: [AP][API][DEBUG] Manage DEBUG status through MQTT commands

ralcaide, 2018-10-25:
Now is only possible to enable / disable DEBUG in compilation time. To add a debug variable into the context, allowing modify the level of debug  from using MQTT commands

[Manage Debug Levels](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/MQTT-Sinapse-API#set-debug-status-enable--disable)

The level of of debug after this task can be 0 (disabled) or any number > 0 (enabled). In other tasks will be developed the different levels of debug

ralcaide, 2018-11-26:
@fpgamcu , what is the status of this task? We are waiting to the implementation

soporteHW, 2019-01-21:
Tested here:

[Set DEBUG status](https://sinapseenergia.testrail.io/index.php?/tests/view/1453)

closed 2019-01-25 by soporteHW