Sinapse-Energia/AP-Sinapse #184: [M95] Limit number of connections per hour and day

ralcaide, 2018-11-09:
In some real installations we are having consumption problems. One of the reasons is that some devices try to connect to the network a lot (maybe) maybe due to a coverage problems or some other problems. As a hot fix we would like to limit the number of connections per hour and per day when GPRS is used. 

The maximum number of connections should be easy to configure in the Definitions.h. Normally we will use something like:

MAX_CONNECTIONS_HOUR = 10
MAX_CONNECTIONS_DAY = 50

ralcaide, 2018-11-09:
@fpgamcu , the consumption problem could be a huge problem for us. Please, work on this with high priority.

If you have any idea about improvements, please share

fpgamcu, 2018-11-09:
Estimation is 4 hours. Commit will be ready today.

ralcaide, 2018-11-09:
great, thanks!

ralcaide, 2018-11-12:
Thanks, should this behavior be implemented also in Bootloader or there is already limited the number of connections?

fpgamcu, 2018-11-12:
It is limiting number of connections in client. I can also implement in bootloader if needed.

ralcaide, 2018-11-12:
Let me think about that. One question: Is there already a limitation mechanism in the BL, isn't? Or if the BL is not able to connect or download the file stays forever in BL? I think no, after several tries go to the client, isn't? 
Please clarify this, if there is already a limitation mechanism we can avoid a new mechanism

fpgamcu, 2018-11-12:
Yes, there is limitation mechanism in the BL. It just not synced up with current implemented in FW, but I think it is not needed, because BL is executed only once at device start / restart.

ralcaide, 2018-11-12:
Great, we will test the current version and if a modification is required in the BL I will let you know through a new issue in BL

soporteHW, 2018-11-15:
With last FW version V1.46.1 the device doesn't connect to MQTT Broker (GPRS model).

Attached the DEV.TXT file saved in SD memory.
[DEV.TXT](https://github.com/Sinapse-Energia/AP-Sinapse/files/2585220/DEV.TXT)

soporteHW, 2018-11-15:
With ETH model, MQTT connection is stablished

[image]

ralcaide, 2018-11-15:
Because the ETH model is still connecting, seems the problem is focused on the new GPRS features.
We are testing with Bootloader 1.30 without deleting the SD. Is maybe related with that? 
@fpgamcu , please take a look to this problem with high priority because at this moment the GPRS version is not working anymore

soporteHW, 2018-11-15:
@ralcaide The SD is erased. The steps performed were:

1) DUT connected with previous FW version (v1.34 or v1.40.1)
2) Sended 207; command to search for nee FW file .bin
3) Sended 208; to activate BL
4) Sended 314; to erase the NVM and restart the device

fpgamcu, 2018-11-16:
GPRS version is working ok in my device.
I will check now bootloader and test with it.

ralcaide, 2018-11-16:
@fpgamcu , in our devices is also working. It was a problem with the binary.

soporteHW, 2018-11-29:
Tested with the following results:

- Max_connections_hour= 10

[image]
[image]

Device works as expected. After 10 reboots (sending "398;0;", at <28-11-2018 16:42:39> device doesn't establish MQTT connection till the next hour. At <28-11-2018 17:01:13>, device establishes MQTT connection again.

- Max_connections_day = 50

[image]

Device works as expected. After 50 reboots (sending "398;0;"), at <28-11-2018 21:24:15> device doesn't establish MQTT connection till the next day.

ralcaide, 2019-03-12:
Tested OK

closed 2019-03-12 by ralcaide