Sinapse-Energia/AP-Sinapse #186: [AQP][ALL][PM] The first value returned is NOT Correct

ralcaide, 2018-12-12:
We observed that when we ask several times to the sensor, the response is very low and not valid. The next responses are OK.

See 
[image]

After read the honeywell DS attached in #160 we see that the response time of the sensor is < 6sg. Maybe it is necessary to:

1. Start the sensor
2. Wait 6s
3. Read the sensor

The goal is to have the most accurate values as possible. 2 consecutive measurements with seconds of difference should return almost the same value

ralcaide, 2018-12-12:
@fpgamcu , what do you think about this problem? Easy to fix? Estimation?

fpgamcu, 2018-12-12:
In our code now it is waiting 6 sec. It is easy to repeat bug and should be solvable in not more than 4-6 hours.

ralcaide, 2018-12-12:
If we are already waiting the 6 sec the problem should be located elsewhere. Please, take a look and let me know