Sinapse-Energia/AP-Sinapse #187: [HW][S2-LP] To connect expansion board to EasyNet

ralcaide, 2018-12-13:
To connect https://www.st.com/en/ecosystems/x-nucleo-s2868a1.html to the current EasyNet in order to have a 6LoWPAN transceiver for Southbound communication

sinapseenergia, 2018-12-14:
[image]

This image shows the connection between X-NUCLEO-S2868A1 board and one EasyNetV2p0 EN_SA_v2p0 board.

sinapseenergia, 2018-12-14:
After connecting the SPI interface, it is needed to enable power with PD3 or ONOFF_Peripheral_1 signal. One generic GPIO is used also for ruling the shutdown in S2-LP.

vAnArhist, 2018-12-26:
From ST Middleware pins GPIO0-GPIO3 is using also for interrupts for MCU in order to notify it about various events. Pin PC7 is used also (see https://www.st.com/content/ccc/resource/technical/layouts_and_diagrams/schematic_pack/group0/2c/a2/0f/06/b6/54/42/de/X-NUCLEO-S2868A1_SCHEMATIC/files/x-nucleo-s2868a1_schematic.pdf/jcr:content/translations/en.x-nucleo-s2868a1_schematic.pdf)

ralcaide, 2018-12-27:
I understand from your comment that several more connections different to the proposed ones should be done, isn't?. 
@vAnArhist , please clarify here, after your analysis, what connections exactly should be done between AP and ST board

vAnArhist, 2018-12-27:
PE6 - GPIO0,
PE7- GPIO1,
PE8 - GPIO2,
PE9 - GPIO3.
Other pin connections is ok.

ralcaide, 2019-01-24:
Last image differs from your previous comment, In the image I only see:

PE9 - GPIO0

Is that right?

fpgamcu, 2019-01-24:
yes, used code from node it is possible to limit connection only to GPIO0 instead of all 4 GPIOs

ralcaide, 2019-03-11:
Tested OK

ralcaide, 2019-06-17:
In the new version of the HW (V3p0) we are using PE6 instead PE9. 

Attached image of the V3p0 design.
[image]

@fpgamcu , could you please modify the FW in order to change this configuration and allow us testing the new devices?

ralcaide, 2019-06-17:
V2p0 : Works OK
V3p0: It does not work

ralcaide, 2019-06-17:
thanks, we will test and provide feedback

closed 2019-07-09 by ralcaide