Sinapse-Energia/AP-Sinapse #188: [S2-LP][Driver][ER] To integrate / develop driver for S2-LP with 6LoWPAN protocol

ralcaide, 2018-12-13:
The current EN is using Sinapse RF in the Southbound to communicate with End Nodes. 
In this new version, the communication with the End Nodes will be done using 6LowPAN (S2-LP). The FW provided by ST should be integrated / adapted in order to provide 6LoWPAN connectivity to EasyNet, converting the current Gateway into a 6LoWPAN Edge Router (MQTT - 6LoWPAN)

Here the FW provided by ST: https://www.st.com/content/st_com/en/products/embedded-software/mcus-embedded-software/stm32-embedded-software/stm32-ode-function-pack-sw/fp-sns-6lpnode1.html

ralcaide, 2018-12-17:
Please, study the FW provided by ST and provide feedback about which one will be used. 
The payload containing the orders to manage the lights will be text messages packaged in UDP messages. The text messages will be defined soon

ralcaide, 2018-12-17:
Attached guide explains how to use the subGhZ FW provided by ST. We should use several parts provided:

- 6LoWPAN stack
- EdgeRouter for Gateway
- EndNode for nodes
- UPD sender/receiver

[ST_FW_4_subGhZ.pdf](https://github.com/Sinapse-Energia/AP-Sinapse/files/2685693/ST_FW_4_subGhZ.pdf)

vAnArhist, 2018-12-18:
Both of those middleware requires Contiki OS and Contiki 6LoWPAN protocol stack 3.x. The main difference from them that in x-cube-subg1 there is examlple of application which use UDP (sender/receiver), when in fp-sns-6lpnode1 there isn`t.
We will try to use x-cube-subg1 (link 2).

ralcaide, 2018-12-18:
ok, thank you for the feeedback.  Questions:

- In the Gateway we are already using FreeRTOS. Will be possible to use Contiki OS or is it possible to use 6LoWPAN protocol stack without Contiki OS?
- In the End Node there is not SO. It is bare metal. How will we proceed?

ralcaide, 2018-12-18:
but is possible to run Contiki OS and FreeRTOS at the same time (Gateway)?
For the End Node, if there is enough memory, will be not a problem to use Contiki

vAnArhist, 2018-12-18:
No, it is not possible to run two OS at the same time.

vAnArhist, 2018-12-18:
https://github.com/hso-esk/emb6

ralcaide, 2018-12-27:
I see the proposed library (without OS) is based in Contiki but I'm afraid maybe is not as reliable as Contiki. We need to have a first version working asap in order to decide if we continue with that or we change to Contiki.
@vAnArhist, @fpgamcu, we need to take a decision by the 27.12.18. Please provide feedback at the end of the day

fpgamcu, 2018-12-28:
After yesterday integration attempt found that emb6 is slightly harder to integrate due to absence of significant stm32 support and dependency on additional build system scons.
So we integrated x-cube-subg1 border-router into our project as most supported by stm32.
DONE: 
integrated transceiver library, nucleo shield library, contiki
project compiles successfully
TODO:
pins configuration (just change BSP header file integrated from x-cube-subg1
contiki run as freertos thread or migrate freertos into contiki
at first we will try to run contiki as freertos thread, if it will fail project will be migrated into contiki
as we checked tonight contiki is required in any case as it contains important libraries to handle ipv6

fpgamcu, 2018-12-28:
Today we will done pins and threads/tasks integration so border router will work in AP project.
How should be the test done? We have now 2 of x-nucleo-s2868a1 boards. Will it be enough to set up one of the x-nucleo-s2868a1 with EN and second with nucleo devboard as node and communicate node with border router?

ralcaide, 2018-12-28:
Good. Yes for a preliminary test the proposed scenario will be fine. Like that at least we are able to check the communication is performed.
Next week (03.01) will arrive to your offices the 6LoWPAN commboards for the NEMA Boards that you already have. Like that you will be able to setup real nodes and a network with 4-5 end nodes should be managed

fpgamcu, 2018-12-28:
Ok.
Most likely for today test we will stay with contiki+freertos, but after we should switch AP totally to contiki to avoid possible multios bugs. Good news that it is not very complicated to switch one rtos to other, so we will try to do this in weekend.

ralcaide, 2018-12-28:
I don't understand totally the strategy, could please verify?

1. Run contiki as a freertos thread
2. Migrate current freertos to contiki and like that AP will run ONLY contiki

Is that right? Why is 1. necessary?
Additional question: Is maybe a good idea to have two different versions of AP? Two branches, one with Contiki and another one with FreeRTOS? I'm afraid Contiki will not cover the features that are already well working in FreeRTOS

ralcaide, 2018-12-28:
If 1. works will be still necessary to implement 2.? I understand yes, as an improvement for stability but maybe not required for the demo, right?

fpgamcu, 2019-01-02:
Hi, we have worked on the border router ST example for Getaway and UDP sender ST example for Node. As we tested on ST boards pure examples work ok, but border router designed to tunnel messages using PC connected to the board. Today we are modifying Border router ST example to tunnel UDP messages from nodet to MQTT.

ralcaide, 2019-03-11:
TESTED OK

ralcaide, 2019-03-25:
@fpgamcu , what frequency are we using? 868Mhz or 915Mhz? If necessary is possible to change it in the FW in a easy way?

closed 2019-07-09 by ralcaide