Sinapse-Energia/AP-Sinapse #189: [6LoWPAN][Lighting][Actuation] To develop the actuation commands for lighting control

ralcaide, 2018-12-13:
It should be developed the commands to act over End Nodes managing lights. These commands will be a mapping / adaptation of the current Sinapse RF Commands:

- TURN ON
- TURN OFF
- Dimming

fpgamcu, 2019-01-28:
To command 1; and 4; assigned node command 03
To command 2; and 3; assigned node command 02
To command 128; assigned node command 01

ralcaide, 2019-02-08:
Testing results were almost OK but have been found two problems bugs:

- The commands 2; and 3; are sending wrong values in order to set some specific dimming values. 

[image]

You can see there that 3;70; 3;60; and 3;30; are setting wrong values. Same problem appears in the AP RF but we tested directly with End Nodes RF and a transparent gateway and the regulation dimming is working well. Conclusion: APs in 6LoWPAN and RF are sending wrong values setting the dimming.

See points 2.1 (6LoWPAN), 2.2 (6LoWPAN) and 2.3 (Legacy EC) of attached document 

- The metering of End Nodes is wrong when the power of the lamp is < than 50W and the dimming is under 40%. With a lamp of 70W is almost OK, with a lamp of 40W. 

See points 3.1 (6LoWPAN 70W), 3.2 (6LoWPAN 40W - ERRORS), 3.3 (Legacy EC) 

See attached document (in Spanish) but you can understand the images

@fpgamcu : Could you please try to fix problems in 6LoWPAN ER - EN. I will create another tasks for AP RF and legacy EC

ralcaide, 2019-02-08:
Test results in Spanish. I think you can understand images. If not, we will translate it to english

[Informe_Tecnico_Test_ DimmingMetering.pdf](https://github.com/Sinapse-Energia/AP-Sinapse/files/2845328/Informe_Tecnico_Test_.DimmingMetering.pdf)

ralcaide, 2019-02-14:
@fpgamcu , any feedback about that?. We need an urgent fix on that.

ralcaide, 2019-02-14:
1. 0..10 Dimming is not working good.
2. Metering in End Nodes when nominal power is less than 50W is wrong. 

We need a fixes on that asap.

fpgamcu, 2019-02-15:
I am now finishing the fixes, should be done today

fpgamcu, 2019-02-18:
DIM conversion fixed, please test it if possible on your side too.

Limitation: table in node used for DIM conversion have 10 points without approximation between them, so dim granularity is limited by tenth of percent. For example 0,10,20....100 should give correct values, while 11, 33, etc. will give incorrect values. I think could be fixed in AP instead of node, if you need such values 11, 33, 53, to be working too I will add an fix.

Now working on metering.

ralcaide, 2019-03-11:
How works now if we sent 3;35; for example? get the value of 30 or 40 or make a lineal conversion?

fpgamcu, 2019-03-18:
In node code it made conversion only if it is exact values of 30 or 40, no linear approximation.
Worse that if you will send 35 node will not convert it in any way and send directly to PWM output.
If node could not be fixed, linear approximation could be applied in getaway side.

ralcaide, 2019-03-29:
Regarding the linkage you have done between EN and ER I have a remark:

> To command 1; and 4; assigned node command 03

Here is applied the legacy formula to convert VV e II to the expected Voltage and Current?
The legacy conversion formula will not be necessary anymore because the FW ER will work only with 6LoWPAN nodes with other STPM10 metering mechanism so the conversion formula should be adapted to new calibration. 
Also, there is a new planed command 04 to get extended metering with PF. It is necessary to think about what command to use and if it is necessary to use the two empty bytes of 03 to provided more data about VV and II having two bytes for both magnitudes instead one byte

> To command 2; and 3; assigned node command 02
> To command 128; assigned node command 01
@fpgamcu , what do you think?

ralcaide, 2019-04-10:
Testing with new metering calibration is not sending correct values when the dimming is > than 90% or when is OFF:

Calibration: https://github.com/Sinapse-Energia/EC-LC-NEMA/issues/29

Explanation of the behavior:
- 316;770004;40;
- 3;0; - > 100;D45BBC;770004;0;0;0;-4.0;234.3;-1.0;0.0;0;14.9;14;0.0;0;1554896976;
- 3;20; - > 100;D45BBC;770004;0;1;22;39.1;234.3;9.0;0.0;0;15.1;14;0.0;0;1554896900;
- 3;60; - > 100;D45BBC;770004;0;1;61;105.0;234.3;24.6;0.0;0;14.7;14;0.0;0;1554897067;
- 3;80; -> 100;D45BBC;770004;0;1;74;127.0;234.3;29.6;0.0;0;14.8;14;0.0;0;1554897104;
- 3;100; - > 100;D45BBC;770004;0;1;87;149.0;234.3;34.8;0.0;0;15.2;14;0.0;0;1554897152;

Here image displaying the how works the formulas in the End Node:

[image]

The ER receives the "result". Maybe is necessary to modify the conversion formulas, isn't?

closed 2019-07-09 by ralcaide