Sinapse-Energia/AP-Sinapse #190: [6LoWPAN][Addressing] To develop the discover protocol in order to be able to communicate with End Nodes

ralcaide, 2018-12-13:
The Gateway (Edger Router - ER) and the End Nodes should implement a discover protocol in order to be able to connect between them and ease the management of the network. 

Normally, this protocol is already included in the driver provided by ST ( #188) but should be correctly used

ralcaide, 2018-12-26:
Here are explained the particularities required by 6LoWPAN Sinapse: https://github.com/Sinapse-Energia/AP-Sinapse/wiki/6LoWPAN-peculiarities

ralcaide, 2018-12-27:
@vAnArhist , I see in the contiki project there are two RPL implementation, the classic one (allowing storage mode) and the little one (more efficient but not allowing the storage mode). Regarding that for the first version will be necessary to communicate only with 5 End Nodes but the real product should allows networks of several thousands devices (up to 30.000), which RPL implementation will we use?

vAnArhist, 2019-01-11:
One address for device in ipv6 routing table uses 16 bytes. 
As we want 30 000 nodes (16*30000 = 480kBytes of ROM). So we cannot use storage mode.
In new releases of Contiki used Lite version without storage. I think we will use lite.

ralcaide, 2019-01-11:
The gateways have SD card. Could we use it to save the addresses, isn't?

fpgamcu, 2019-01-28:
After commit 53dcf97 address scheme in ER is changed.
Now using last 3 bytes of REG2 as node ID.
So, in node its ID registers should be REG0=0x01234567, REG1=0x89ABCDEF, REG2=0x02XXXXXX, where XXXXXX is any hexadecimal number identifying the node.
When creating node with 316; command in ER, XXXXXX should be used as node ID. ER will automatically perform conversion of this ID into node IPv6 address to communicate with node.

ralcaide, 2019-03-11:
TESTED OK

closed 2019-03-11 by ralcaide

ralcaide, 2019-06-26:
Broadcast address and group address is are not developed