Sinapse-Energia/AP-Sinapse #191: [RF RX][Responses Management] Questions & Improvements

ralcaide, 2018-12-17:
After perform several very specific tests, we have several questions:

1. Regarding metering, command 1; - 101;, we see the data & timestamp is only updated when the AP receives **two** responses and not only **one**. This is not a valid behaviour, the updated 101; data should be updated each time a response is received. 

2. Are the collisions managed somehow? Several EPDs can transmit at the same time and maybe the messages arrives at the same time. There is a mechanism to manage this situation?

ralcaide, 2018-12-17:
@fpgamcu , could answer both questions?. The question one should be fixed if it is not working as expected. The second one is not necessary by the moment

fpgamcu, 2018-12-18:
1. I will check and fix it today.
2. No, there is no collision management.

ralcaide, 2019-03-12:
@fpgamcu , what is the status of this task?

fpgamcu, 2019-03-18:
Checked p.1 - it need some time for debug and found an issue, and also test with new drivers.
I think 16 hours more should be enough.

ralcaide, 2019-07-09:
Now focused on 6LP. RF Sinapse will be maintained if needed but with low priority